zephyrproject-rtos/zephyr #94006: Sysbuild sample with_mcuboot for xiao_ble board fails with linker errors

sidcha, 2025-08-01:
### Describe the bug

I'm trying to build the sysbuild sample `with_mcuboot` for the `xiao_ble` board with:

```
west build -p auto -b xiao_ble --sysbuild samples/sysbuild/with_mcuboot
```

That fails with (full build logs [here](https://pastebin.com/raw/wqejYndG)):
```
CMake Error at /home/sidcha/work/zephyrproject/zephyr/cmake/modules/extensions.cmake:4107 (message):
  dt_prop(slot0_size ...) missing required argument: PATH
Call Stack (most recent call first):
  CMakeLists.txt:471 (dt_prop)
```

Taking a closer look, I discovered that the board definition assumes a UF2 build and did not have the required partitions for mcuboot. With the following change, I was able to make progress.

```diff
diff --git a/boards/seeed/xiao_ble/xiao_ble_common.dtsi b/boards/seeed/xiao_ble/xiao_ble_common.dtsi
index 47c3b300ae1..a7f40a874a6 100644
--- a/boards/seeed/xiao_ble/xiao_ble_common.dtsi
+++ b/boards/seeed/xiao_ble/xiao_ble_common.dtsi
@@ -6,7 +6,7 @@
  */
 
 #include <nordic/nrf52840_qiaa.dtsi>
-#include <nordic/nrf52840_partition_uf2_sdv7.dtsi>
+#include <nordic/nrf52840_partition.dtsi>
 #include "xiao_ble-pinctrl.dtsi"
 #include "seeed_xiao_connector.dtsi"
 
diff --git a/boards/seeed/xiao_ble/xiao_ble_defconfig b/boards/seeed/xiao_ble/xiao_ble_defconfig
index 725c581b364..f6439c1e67d 100644
--- a/boards/seeed/xiao_ble/xiao_ble_defconfig
+++ b/boards/seeed/xiao_ble/xiao_ble_defconfig
@@ -14,7 +14,3 @@ CONFIG_SERIAL=y
 
 # Enable console
 CONFIG_CONSOLE=y
-
-# Build UF2 by default, supported by the Adafruit nRF52 Bootloader
-CONFIG_BUILD_OUTPUT_UF2=y
-CONFIG_USE_DT_CODE_PARTITION=y
```

But the linking zephyr/zephyr_pre0.elf fails (full logs [here](https://pastebin.com/raw/guCqKzwY)). Can someone help?

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

sidcha, 2025-08-02:
Looks like I need this too:

```diff
diff --git a/samples/sysbuild/with_mcuboot/sysbuild/mcuboot.conf b/samples/sysbuild/with_mcuboot/sysbuild/mcuboot.conf
index 3e228ee2ac3..58c39fa8005 100644
--- a/samples/sysbuild/with_mcuboot/sysbuild/mcuboot.conf
+++ b/samples/sysbuild/with_mcuboot/sysbuild/mcuboot.conf
@@ -3,3 +3,4 @@
 # sysbuild will automatically set it for both the application and MCUboot images itself
 CONFIG_MCUBOOT_LOG_LEVEL_WRN=y
 CONFIG_MCUBOOT_DOWNGRADE_PREVENTION=y
+CONFIG_MULTITHREADING=y
```

With that, the build proceeds but I run out of flash space:

```
/home/sidcha/zephyr-sdk-0.17.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
/home/sidcha/zephyr-sdk-0.17.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 1408 bytes
collect2: error: ld returned 1 exit status
```

prashant3285, 2025-10-22:
>adjusting the partition layout from a dts board overlay with /delete-node/ ... entries.

can you share the partition layout changes