zephyrproject-rtos/zephyr #94176: drivers: clock_control: Nordic nRF54L15 HFCLK can't be activated with `clock_control_on()` anymore

GardeningStevie, 2025-08-06:
### Describe the bug

Since PR  #86354, my application can't activate the HF clock source (external HF crystal) with `clock_control_on()` when the crystal is physically off. The error code `-EPERM` (-1) is returned.

Actually, the internal `flags` state is set to `2` which means `CLOCK_CONTROL_STATUS_ON`. That's the reason why `clock_control_on()` returns -1. However, the clock control state is wrong. I have checked the CLOCK register and crystal itself. It is off.

The reason for `CLOCK_CONTROL_STATUS_ON` is the call of `clkstarted_handle()`  here 
https://github.com/zephyrproject-rtos/zephyr/blob/0f66e9c68088e79f33edb5ed500d9f54da6a2c3e/drivers/clock_control/clock_control_nrf.c#L703-L706
`clkstarted_handle()`  enforces that the internal `flags` state is set to `CLOCK_CONTROL_STATUS_ON`. Somehow that state is never reset to `CLOCK_CONTROL_STATUS_OFF`.

I could "fix" this issue when I replaced the current implementation with the old one that checked if the current state is `CLOCK_CONTROL_STATUS_STARTING` which make sense IMHO.
https://github.com/zephyrproject-rtos/zephyr/blob/c326bf6bc718afc72a1cd78ef97c73f95051284d/drivers/clock_control/clock_control_nrf.c#L719-L730
I didn't check if the clock is handled really correctly with this work-around. I have just noticed that with this fix `clock_control_on()` and `clock_control_off()` work as expected.

I use these feature for crystals frequency checking in production. Unable to enable the crystal permanently is a release blocker for me.

### Regression

- [x] This is a regression.

### Steps to reproduce

Not entirely sure what is really required to reproduce. I have ...

1. Bluetooth application in advertising mode
2. Stop BT advertising
3. Call `clock_control_on(CLOCK_DEVICE, CLOCK_CONTROL_NRF_SUBSYS_HF);`

### Relevant log output

```shell

```

### Impact

Showstopper – Prevents release or major functionality; system unusable.

### Environment

- Zephyr 4.2
- Nordic nRF54L15

cvinayak, 2025-12-09:
Relates to #99909?
Relates to #100754?

cvinayak, 2025-12-11:
Thank you @nordic-krch 

@GardeningStevie I found an example from @rick1082 below:

```
+static int clocks_start(void)
+{
+       int err;
+       int res;
+       struct onoff_manager *clk_mgr;
+       struct onoff_client clk_cli;
+
+       clk_mgr = z_nrf_clock_control_get_onoff(CLOCK_CONTROL_NRF_SUBSYS_HF);
+       if (!clk_mgr) {
+               printf("Unable to get the Clock manager\n");
+               return -ENXIO;
+       }
+
+       sys_notify_init_spinwait(&clk_cli.notify);
+
+       err = onoff_request(clk_mgr, &clk_cli);
+       if (err < 0) {
+               printf("Clock request failed: %d\n", err);
+               return err;
+       }
+
+       do {
+               err = sys_notify_fetch_result(&clk_cli.notify, &res);
+               if (!err && res) {
+                       printf("Clock could not be started: %d\n", res);
+                       return res;
+               }
+       } while (err);
+
+       /* TODO: Take care of any anomaly workaround as required */
+
+       printf("HF clock started\n");
+
+       return 0;
+}

```

If this helps and satisfactory, please close the issue.