zephyrproject-rtos/zephyr #94181: Busfault in http_server_code in case if famuly is AF_UNSPEC

AndreyDodonov-EH, 2025-08-06:
### Describe the bug

Target: any

After I added custom socket creation for the webserver, https://github.com/zephyrproject-rtos/zephyr/pull/92732,
it is possible to set any kind of `sa_family_t` for the used socket.

E.g. in case of my offloaded soket I set `AF_UNSPEC` since it's a direct link without any IP addresses.
I assume others might do the same in the future.

Then we call https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/http/http_server_core.c#L279
```
LOG_DBG("New client from %s:%d",
		net_sprint_addr(sa.ss_family, &net_sin((struct sockaddr *)&sa)->sin_addr),
		ntohs(net_sin((struct sockaddr *)&sa)->sin_port));
```

`net_sprint_addr` returns `NULL` for `AF_UNSPEC` and calling `LOG_DBG` with `NULL` for `%s` leads to a busfault.

Potential fixes:
1. Make `z_impl_net_addr_ntop` return not `NULL`
2. Add a NULL-check before calling `LOG_DBG` or family check before calling `net_sprint_addr`
3. Somehow forbid `AF_UNSPEC`, but I don't think it's a good idea

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Create HTTP Service with a custom config which creates a socket of family `AF_UNSPEC`

```
static struct http_service_config serial_http_service_config = {.socket_create =
                                                                    http_service_socket_create};

HTTP_SERVICE_DEFINE(http_service, NULL, &http_service_port, MAX_CONCURRENT_CONNECTIONS,
                    MAX_BACKLOG_CONNECTIONS, NULL, NULL, &serial_http_service_config);

...
static int __maybe_unused http_service_socket_create(const struct http_service_desc *svc, int af,
                                                     int proto)
{
    ...
    int fd = zsock_socket(AF_UNSPEC, SOCK_STREAM, 0);
    ...
```

2. Set `CONFIG_NET_HTTP_SERVER_LOG_LEVEL_DBG=y`
3. Start the program, observer the error

### Relevant log output

```shell
[00:00:02.899,000] <err> os: bus_fault: ***** BUS FAULT *****
[00:00:02.899,000] <err> os: bus_fault:   Precise data bus error
[00:00:02.899,000] <err> os: bus_fault:   BFAR Address: 0x0
[00:00:02.899,000] <err> os: esf_dump: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x00000002
[00:00:02.899,000] <err> os: esf_dump: r3/a4:  0x00000000 r12/ip:  0x00000001 r14/lr:  0x080073cb
[00:00:02.899,000] <err> os: esf_dump:  xpsr:  0x21000000
[00:00:02.899,000] <err> os: esf_dump: Faulting instruction address (r15/pc): 0x08001382
[00:00:02.899,000] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:02.899,000] <err> os: z_fatal_error: Current thread: 0x20005320 (http_server_tid)
[00:00:02.906,000] <err> os: k_sys_fatal_error_handler: Halting system
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Any version after https://github.com/zephyrproject-rtos/zephyr/pull/92732 got merged
- Environment-independent

AndreyDodonov-EH, 2025-08-06:
@rlubos @cfriedt 
Would like you recommendation on this one

cfriedt, 2025-08-06:
@AndreyDodonov-EH - I think it's likely that the log call

```cpp
LOG_DBG("New client from %s:%d",
		net_sprint_addr(sa.ss_family, &net_sin((struct sockaddr *)&sa)->sin_addr),
		ntohs(net_sin((struct sockaddr *)&sa)->sin_port));
```

assumed that the socket had family `AF_INET`.

An alternative that should hopefully also work is for the http server to abstract stringifying a socket (with some stack memory?)

Would you be up for making a PR?

AndreyDodonov-EH, 2025-08-07:
@cfriedt 
Yes, exactly, previously that assumption was correct.
I think it's a little bit of an overkill to add hier extra layer of abstraction (and consume stack which might be a breaking change for some).
If that's okay for you, I'd just add a change which @rlubos suggested

closed 2025-08-11 by cfriedt