zephyrproject-rtos/zephyr #94221: Missing timers on STM32N6 device trees

EBouillot, 2025-08-07:
### Describe the bug

Trying to use quadrature decodeur and pwm on a nucleo_n657x0_q, I found no declaration of the timers (TIM) on dtsi and dts (unlike other nucleo boards, where those are declared on the provided device trees).
PS : I concede being unexperimented using Zephyr and manipulating device trees, so I might have missing something

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

rruuaanng, 2025-08-09:
@EBouillot You can try adding this:
```
diff --git a/boards/st/nucleo_n657x0_q/nucleo_n657x0_q_common.dtsi b/boards/st/nucleo_n657x0_q/nucleo_n657x0_q_common.dtsi
index 4232538817f..8c2c5e6cfd3 100644
--- a/boards/st/nucleo_n657x0_q/nucleo_n657x0_q_common.dtsi
+++ b/boards/st/nucleo_n657x0_q/nucleo_n657x0_q_common.dtsi
@@ -177,6 +177,16 @@
        status = "okay";
 };
 
+&timers1 {
+       status = "okay";
+
+       pwm1: pwm {
+               status = "okay";
+               pinctrl-0 = <&tim1_ch1_pa8>;
+               pinctrl-names = "default";
+       }
+}
+
 &usart1 {
        clocks = <&rcc STM32_CLOCK(APB2, 4)>,
                 <&rcc STM32_SRC_CKPER USART1_SEL(1)>;
```

EBouillot, 2025-08-11:
> [@EBouillot](https://github.com/EBouillot) You can try adding this:
>
> ```
> diff --git a/boards/st/nucleo_n657x0_q/nucleo_n657x0_q_common.dtsi b/boards/st/nucleo_n657x0_q/nucleo_n657x0_q_common.dtsi [...]

I already tried that, but failed with error "undefined node timers1"