zephyrproject-rtos/zephyr #94268: hal_stm32: STM32H7S3L8 PinCtrl Generation does not provide Ethernet

UmbraMalison, 2025-08-08:
### Describe the bug

genpinctrl.py does not identify the Ethernet pins for the H7S3L8. 

See -> https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/h7/stm32h7s3l8hx-pinctrl.dtsi#L1112

Some pins are generated, seemingly only the ones defined simply as "ETH_" such as "ETH_MDIO" where as those defined differently to other H7s such as "ETH_RMII_TXD0" AND "ETH_MII_TXD0" are not. I have tried changing the names in the open pin repo but the results were the same.

Ethernet pins are provided in -> https://github.com/STMicroelectronics/STM32_open_pin_data/blob/master/mcu/STM32H7S3L8Hx.xml

I've taken a look at the script, tried a few things like ensuring it can't use the F1 config files where extended ETH expressions are not defined. Nothing I have tried so far has had any effect on the output.

Happy to try anyones ideas :)

Based on the impact descriptions, this prevents ethernet from being usable which is a significant function of the board, and the workaround for this is certainly difficult IMO if you want to retain zephyr API and device tree.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. mkdir pinctrltest
2. git clone https://github.com/STMicroelectronics/STM32_open_pin_data.git
3. git clone https://github.com/zephyrproject-rtos/hal_stm32.git
4. python3 hal_stm32/scripts/genpinctrl/genpinctrl.py -p STM32_open_pin_data/ -o ./out
5. review h7s3l8 for ethernet pins

### Relevant log output

```shell
all relevant outputs are in upstream as of now.

https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/h7/stm32h7s3l8hx-pinctrl.dtsi#L1112
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

Linux FC 42 using devcontainer zephyrprojectrtos/zephyr-build:latest
Zephyr v4.2.0
SDK 17.3

MY201314MY, 2025-08-20:
there's no ethernet peripherals in the device tree under zephyr repository.
Could anyone please consider adding them?
Should be in **_zephyr/dts/arm/st/h7rs/stm32h7rs.dtsi_**

mariopaja, 2025-08-22:
@UmbraMalison @MY201314MY 

I have the missing pinctrl and nodes already since I tried once to enable ethernet on H7RS.

However, I faced also some issues on the driver side, which I didn't chase after.

At this point, I would not start working on it because there is also an ongoing effort to split the v1 from v2 drivers https://github.com/zephyrproject-rtos/zephyr/pull/94142

At this point I think it should be be changed from bug to enhancement.

mariopaja, 2025-09-02:
> With PR https://github.com/zephyrproject-rtos/hal_stm32/pull/297 merged, I propose to close this ticket.

@djiatsaf-st for me it is fine. After all it is not a bug :D

closed 2025-09-02 by djiatsaf-st