zephyrproject-rtos/zephyr #94282: `spi_transceive_cb()` function is causes an infinite loop on STM32 processors

jpowen898, 2025-08-08:
### Describe the bug

I am using the `stm32h7` and attempting to use `spi_transceive_cb()` function with `CONFIG_SPI_STM32_INTERRUPT` and `CONFIG_SPI_ASYNC` selected. Calling this function causes the processor to freeze permanently and never recover. When debugging this code I found that it gets stuck in this do while loop forever. 

https://github.com/zephyrproject-rtos/zephyr/blob/dece77924f7c415691b1112ac38fc796d05dd640/drivers/spi/spi_ll_stm32.c#L1023-L1031

I'm not sure I understand exactly how `spi_context_wait_for_completion()` is intended to be used, but when you are in `ctx.asynchronous` mode that function returns immediately instead of blocking and with a return code of 0. This is a problem because the STM32 SPI driver is waiting for it to return something non 0 so it gets stuck forever. Also I would think the `spi_stm32_transfer_ongoing(data)` function would keep it from blocking forever, but I looked at this with a logic analyzer and the transaction never actually starts. The SPI bus setup happens (changes CLK to idle HIGH, and sets CS LOW) but then nothing else happens and it gets stuck forever.

I fixed it for my application by changing the loop condition so that the loop exits immediately when the transaction is supposed to be async. This allows the `spi_transceive_cb()` to return imediately and then the `spi_cb()` function I passed it gets called later at the end of the spi transaction as expected.
```
  } while (!ret && spi_stm32_transfer_ongoing(data) && !data->ctx.asynchronous);
```

### Regression

- [x] This is a regression.

### Steps to reproduce

Use the `nucleo_h743` board, setup your SPI device and call `spi_transceive_cb()`. This function never returns and it gets stuck in a infinite loop.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

Linux OS using zephyr v4.2.0 release and the v0.17.2 SDK

erwango, 2025-11-03:
Under review

jpowen898, 2026-01-17:
This problem is fixed by #94690 but it looks like it just got closed without getting merged...