zephyrproject-rtos/zephyr #94285: arm64: Interrupt nesting breaks when FP operations are used in ISR

keith-packard, 2025-08-08:
### Describe the bug

arm64 is supposed to support interrupt nesting, but if any FP/SIMD operations are performed during interrupt processing, interrupts are disabled for the remainder of the ISR.

This is because the only place to save FPU state is in the task struct. One option would be to allocate space in the interrupt stack alongside the regular registers so that each interrupt could save FPU context separately.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Steps to reproduce:
 1. Apply this patch:
```
diff --git a/tests/kernel/interrupt/src/nested_irq.c b/tests/kernel/interrupt/src/nested_irq.c
index 9c9d909b546..7c78fc9e90b 100644
--- a/tests/kernel/interrupt/src/nested_irq.c
+++ b/tests/kernel/interrupt/src/nested_irq.c
@@ -113,8 +113,10 @@ void isr1(const void *param)
 void isr0(const void *param)
 {
        ARG_UNUSED(param);
+       volatile float one = 1.0;
+       volatile float two = 2.0;
 
-       printk("isr0: Enter\n");
+       printk("isr0: Enter %d\n", (int) (two * one));
 
        /* Set verification token */
        isr0_result = ISR0_TOKEN;
```
 2. west build -p -b qemu_cortex_a53 -d build-18 tests/kernel/interrupt -T arch.interrupt
 3. run the test

### Relevant log output

```shell
START - test_nested_isr
Triggering irq : 14
isr0: Enter 2
Triggering irq : 15

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/interrupt/src/nested_irq.c:131: isr0: (isr1_result not equal to ISR1_TOKEN)
isr1 did not execute
isr0: Leave
isr1: Enter
isr1: Leave
 FAIL - test_nested_isr in 0.006 seconds
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

andyross, 2025-08-08:
Driving by so GitHub sees my name attached and sends updates. 

Agree with the "put it on the stack" notion.  You can take a look at the existing xtensa arch, or the in-progress Cortex M arch_switch() PR for examples that work like this.  Both support nested exceptions naturally, just using whatever stack the interrupt landed on (well, not that simple because ARM has a stack switch so it's not literally pushing to $SP, but that's the principle).

keith-packard, 2026-04-01:
The proposed fix was to reserve space for FPU saving in the interrupt stack but not save until an FPU trap occurred; that consumes space but wouldn't impact performance. It looked like it would be a modest change in the code; just adapting where the FPU save/restore happened. But, as you say, it's not without impact and we have compiler hacks that work around the varargs/memcpy issues. I did get push-back from gcc upstream when proposing that as a 'real' fix though, so something that didn't rely upon that would be nice.

npitre, 2026-04-01:
Reserving space on the interrupt stack is doable, but the cost grows quickly. With NEON alone that's ~520 bytes per nesting level. With SVE it becomes variable and potentially very large depending on `CONFIG_ARM64_SVE_VL_MAX`:

| VL (bits) | Z regs (32×VL) | P regs (16×VL/8) | FFR | Total |
|---|---|---|---|---|
| 128 | 512 B | 32 B | 2 B | ~554 B |
| 256 | 1 KB | 64 B | 4 B | ~1.1 KB |
| 512 | 2 KB | 128 B | 8 B | ~2.2 KB |
| 2048 | 8 KB | 512 B | 32 B | ~8.7 KB |

Not impossible, but it puts significant pressure on interrupt stack sizing, especially with deeper nesting.

It's also worth noting that Zephyr's official documentation states: "The kernel does not support the use of floating point registers by ISRs." On x86 for instance, FPU access from an ISR simply traps with no recovery path. ARM64 (and RISC-V) already go beyond that baseline by allowing it to work at all, with the trade-off of disabling nested IRQs. So the value of doing more here isn't necessarily universal, and heavy FP work in ISRs is probably better delegated to a thread context anyway.

As for the GCC upstream pushback on `-moverride=tune=no_ldp_stp_qregs`, what was the objection?