zephyrproject-rtos/zephyr #94315: RT700 GPIO0 can't work because hifi core supported gpio0 function

lucien-nxp, 2025-08-11:
### Describe the bug

Blinky case supported on cm33_cpu0 can't work because the PCNS/ICNS/PCNP/ICNP value have been modified in board.c
Hash: 40f08428d3e1991d8b0e78fd35bb4daa1cc96653

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

lucien-nxp, 2025-08-11:
Hi @VitekST , could you help check?  I found your commit involve the issue.

VitekST, 2025-08-11:
@lucien-nxp I've done some experimenting and looks like the silicon is wired in such a way that the `ICNS` and `PCNS` registers control accesses from the HiFi 4 domain, the `ICNP` and `PCNP` have no effect in this case. As can be implied from the RM, the possible settings are mutually exclusive, so for GPIO access, it's either the CM33 #0 or HiFi 4 for, at least, GPIO0.

I guess I'll proceed with simply reverting that commit and the enablement of the GPIO0 peripheral from the HiFi 4 domain entirely. Apologies for that, I wasn't as thorough when reading the manual as I should've been.