zephyrproject-rtos/zephyr #94364: IPC and multiprocessor applications in native-sim

dsseng, 2025-08-11:
### Is your feature request related to a problem? Please describe.

Nowadays there is an increasing number of multicore embedded systems. In many cases, the user wants to test interactions between applications running in their solution, and it would be really great to enable native simulator to communicate with another native simulator using virtual IPM/MBOX and a shared memory region.

This would benefit not only users, but Zephyr as well, because IPC backends seem to receive less testing than desired (some of them are likely tested with nrf5340_bsim, but I cannot say for sure).

### Describe the solution you'd like

This might have multiple approaches, but I would try using a Linux shared memory object to back a shared memory region (to simulate what real hardware has), and a UNIX socket to communicate mailbox events. To me, this seems to represent actual implementation in HW really well, allowing to test both IPC backends and custom application protocols over IPC, as well as integration-test the whole solution verifying multiple applications communicate and are able to produce the desired result

### Describe alternatives you've considered

Boil down secondary application to basic functions to be exposed via Zephyr APIs, enabling testing primary CPU on native simulator, then testing secondary app manually on hardware. This creates a suboptimal architecture not fully covered by tests, so I propose an improvement.

aescolar, 2025-08-12:
Hi @dsseng

-------
Background info: 
`native_sim`, the zephyr target board, is a single MCU target (i.e. not AMP).

The `native simulator` and the way it is integrated in Zephyr supports multiMCU/AMP builds.
The nrf5340bsim includes 2 MCU targets (2 Zephyr target boards, nrf5340bsim/nrf5340/cpu{app,net} ), and when built together they can communicate with each other, and do so in quite a few tests we run in CI. Some of the IPC backends and OpenAMP are tested with it.

How this works is that all MCU images are built into the same process, i.e. there is no need for shared memory region, as they share the process memory. Building them in the same process also allows the MCUs and HW models to be scheduled coherently with little overhead.

-------

Wrt. testing the IPC backends, we could check which are (not) tested today in the nrf5340bsim and see if we can do better.

For other uses, it really depends on what one would like.
The `native_sim` target has always tried to be a relatively simple, very HW agnostic test platform. We could consider adding an AMP variant with 2 or more cores apart from the nrf5340bsim if it would be beneficial.
But note that as soon as we add AMP we add a lot of complexity that makes things more difficult for users.

Can you elaborate about what it is that you would like to test ?

dsseng, 2025-08-12:
Hello!

Yes, I was referring to nrf5340_bsim as an example of how a native AMP target is built. These aspects can certainly be expanded to the native-sim.

Yes, probably to test the internal IPC backends we can just add overlays to the tests with nrf5340_bsim. Should I take a look at that?

I think about offloading some operations regarding CAN periodic messages and other nontrivial tasks I want to cover with tests to the second core in my project. This means I would need second core's functions to run to be able to interact with the simulated app for the e2e tests.

We should review the variants and see how to control complexity and keep it contained to the AMP variant (perhaps that would be either a separate target, or we could use dt/Kconfig to add more cores via sysbuild linkage to the existing native-sim-64). What do you think on the complexity? We already have 5340 as the example, but pulling bsim in when the project is nothing to do with nRF or radio, and needs networking features is simply not logical and will make testing hard.

aescolar, 2025-08-12:
> Yes, probably to test the internal IPC backends we can just add overlays to the tests with nrf5340_bsim. Should I take a look at that?

You are welcome to.

>  We already have 5340 as the example, but pulling bsim in when the project is nothing to do with nRF or radio, and needs networking features is simply not logical and will make testing hard.

It is not ideal. But it exists today :)

> This means I would need second core's functions to run to be able to interact with the simulated app for the e2e tests.

You may want to consider, for testing, running that functionality in the same core, just in different threads (and instead of using an inter core mechanism use a plain inter thread mechanism).

> We should review the variants and see how to control complexity and keep it contained to the AMP variant (perhaps that would be either a separate target, or we could use dt/Kconfig to add more cores via sysbuild linkage to the existing native-sim-64). What do you think on the complexity?

It is quite complex. But it is doable if the benefit would justify it. For N cores we need at least N interrupt controllers and N system tick timers, and similarly we need their drivers interacting with the right HW instance. For other native_sim drivers we'd need to be careful to ensure they can play nicely with another instance running in another core (and I already know, today, they will not).
For users, using an AMP target, complexity jumps up immediately, for once they need to use sysbuild (which is not that easy), between other things.

aescolar, 2025-08-12:
> Are there any drawbacks to the separate executables idea?

`native_sim` is as its core deterministic, which makes reproducing and debugging much easier.
Using asynchronous mechanisms for communication between executable would lose that, but in a really bad way (think one executable will run to completion before the other sends a single message or viceversa depending on load. Pausing one executable in a debugger would make everything fail, etc.).
On the other hand running things in different executables while maintaining determinism would require a lot of synchronization which would have a very heavy impact on runtime performance. (Note that babblesim only synchronizes processes when it needs to, which, simplifying, is when a radio packet is exchanged, which is actually quite seldom overall)

What I have seen is that debugging a multi image executable is in general easier than it may seem.

Meaning, I made that design choice very consciously, and I haven't had any reason to regret it so far.

closed 2026-09-09 by github-actions[bot]