zephyrproject-rtos/zephyr #94405: RFC: Breaking API Change: crypto/hash.h: make hash_pkt input buffer constant

titouanc, 2025-08-12:
### Problem Description

Currently in the Zephyr hash API (as of v4.2.0), the input buffer is defined as `uint8_t *`, see https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/include/zephyr/crypto/hash.h#L90-L91

Consequently, feeding a constant input to hash requires to either:
* make a type-unsafe cast `(const uint8_t *) -> (uint8_t *)`, which is const-incorrect
* make a copy of the input in a non-const buffer, and then pass it to the hash function, which is inefficient, and opens the door to buffer overflows or similar issues.

[From discussions with other Zephyr developers](https://github.com/zephyrproject-rtos/zephyr/pull/94218#pullrequestreview-3097064416), it seems that this was initially envisioned to allow for in-place cryptographic operations, where the hashed result would replace the input to be hashed in the same buffer. However, when looking at actual implementations, there does not seem to be any instance of that pattern (this is only used in ciphering operations).

In fact, all hashing backends (either hardware or mbedtls) take a constant input, and write the resulting hash in a separate output buffer.

There is even a sample in-tree (`samples/boards/microchip/mec172xevb_assy6906/rom_api`) that shows this problem:
1. There is a constant input to hash: https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/samples/boards/microchip/mec172xevb_assy6906/rom_api/src/main.c#L409
2. The input must be cast to non-const to fit in the hash API: https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/samples/boards/microchip/mec172xevb_assy6906/rom_api/src/main.c#L444
3. Under the hood, this will then be passed to the actual hash backend as `const` again https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/drivers/crypto/crypto_mchp_xec_symcr.c#L361

### Proposed Change (Summary)

Mark the input buffer as constant (ie. `const uint8_t * in_buf` in `struct hash_pkt`)

### Proposed Change (Detailed)

See proposed changes and initial discussion here: https://github.com/zephyrproject-rtos/zephyr/pull/94218

### Alternatives Considered

Keeping everything as is, and casting a `const uint8_t *` to `uint8_t *` when calling the hash API will not cause any issue, because all backends will actually use it as a `const uint8_t *` under the hood. However, this raises unnecessary compiler warnings.

d3zd3z, 2025-08-15:
I think this is reasonable. As far as how to comply with this, because there needs to be a transition period for an API change, that suggests a feature would be needed to control this, which seems heavy.

Perhaps we could argue: 1. this is a trivial change, and 2. This fixes an incorrect API. But, I'm not sure we have a process to do this any faster.

In general, this shouldn't affect client code, unless that client code is using this pointer itself to write to the buffer, in which case a cast in would need to be added to the code.

ceolin, 2025-09-18:
This looks correct to me. I thought this was done this way because of TinyCrypt (that was the first implementation for this driver) but even there `const` is expected.

closed 2025-10-21 by jhedberg