zephyrproject-rtos/zephyr #94410: Out of bounds array access in DNS dispatcher in zephyr v3.7.1

jbr-ia, 2025-08-12:
### Describe the bug

I am trying to upgrade a program from zephyr v3.7.0 to v3.7.1. 
With v3.7.1, I am having problems with DNS-requests. When a dns-request is done, an array is read out of bounds, and in my case the microcontroller stops with a 'Precise-databus-error'.

This is seems to be the same problem as: https://github.com/zephyrproject-rtos/zephyr/issues/79042
And happens to me in the zephyr v3.7-branch since commit: https://github.com/zephyrproject-rtos/zephyr/pull/78281

In the current main zephyr-version, this is already solved. By cherry-picking commits from: https://github.com/zephyrproject-rtos/zephyr/pull/79056, it seems to work for me with zephyr v3.7.1. 
Can that pull-request also be backported to v3.7?

### Regression

- [x] This is a regression.

### Steps to reproduce

I am testing with a custom board with a custom application, but the method from:  https://github.com/zephyrproject-rtos/zephyr/issues/79042 will probably work.

### Relevant log output

```shell
[00:00:19.254,000] <err> os: ***** BUS FAULT *****
[00:00:19.254,000] <err> os:   Precise data bus error
[00:00:19.254,000] <err> os:   BFAR Address: 0x39
[00:00:19.254,000] <err> os: r0/a1:  0x0000002d  r1/a2:  0x00000006  r2/a3:  0xffffffff
[00:00:19.254,000] <err> os: r3/a4:  0x00000010 r12/ip:  0x00000006 r14/lr:  0x0804ca1b
[00:00:19.254,000] <err> os:  xpsr:  0x21000000
[00:00:19.255,000] <err> os: Faulting instruction address (r15/pc): 0x0805a0d8
[00:00:19.255,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:19.255,000] <err> os: Current thread: 0x20005c50 (net_socket_service)
[00:00:19.256,000] <err> os: Halting system
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

OS: Ubuntu
Toolchain: zephyr-sdk-16.8
Zephyr: v3.7.1

jukkar, 2025-08-12:
I added backport-3.7 label to the PR, lets see if the bot is able to create suitable PR for it. See #94414

jhedberg, 2025-08-12:
@ceolin FYI

carlescufi, 2025-10-01:
Fixed by #94414

closed 2025-10-01 by carlescufi