zephyrproject-rtos/zephyr #94418: [ARM][NXP]FRDM-MCXN947 LPSPI not working with zephyr 4_2_0.

csann, 2025-08-12:
### Describe the bug

I have a simple zephyr program to read and write data via SPI to an ad74413r add/dac device. The app was developed to help me learn zephyr / use of the lpspi / and  and to control the device. It works with zephyr v4_1_0 but does not work with zephyr v4_2_0 using either SDK 0.17.0 or 0.17.2. 

With 4_1_0, when I send a variety of 32 bit commands to the device, CD first drops, at the same time SCK appears. I have it set to 1 MHz. Then 32 bits of data are sent. Then SCK stops and CD rises. I can read and write data to the device and all messages follow this pattern.

[image]

With 4_2_0, the CD, SCK, and data signals all change. SCK changes to 6.25 MHz and the data is sent a byte at a time with long delays between bytes. SCK is changed from a continuous stream to send 32 bits, to a series of 4 pulses of SCK, one for each byte. 

[image]

I will add my overlay file, main.c and screen shots of the logic analyzer showing the difference between 4_1_0 and 4_2_0.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Run app on 4_1_0 with SDK 0.17.0. App works fine. 
Change to zephyr 4_2_0 with SDK 0.17.0. App breaks.
Change SDK to 0.17.2. App remains broken. No change from previous step/

### Relevant log output

```shell

```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

Mac OS Sequoia, latest version, running on either an Intel MacBook Pro or an Apple Silicon MacBook Pro M4 Max. It does not seem to be dependent on the host processor.

Zephyr 4_1_0 works fine. Zephyr 4_2_0 exhibits the problem.

SDK 0.17.1 performs identically on both zephyr versions. The system work with 4_1_0 and does not work on 4_2_0 with either.  SDK 0.17.2 does not fix the problem on 4_2_0. I did not test 0.17.2 with 4_1_0.

csann, 2025-08-31:
I would like to know why this has a low priority since it breaks SPI on at least this very common NXP MCU?

alxrey, 2025-09-01:
I agree with @csann.
There’s no need for a specific application to notice the issue. **The `spi_loopback` test fails immediately** when run on the FRDM-MCXN947 (branch main), which indicates that SPI apparently no longer works on this SoC family. For me, this is a major issue.

alxrey, 2025-10-31:
> I have just tried on the latest main ([82f419d](https://github.com/zephyrproject-rtos/zephyr/commit/82f419de74198e5103c25ab6df0ef3f0a41361a7)). I build the spi loopback test for this platform with the following commands: `west build -b frdm_mcxn947//cpu0 -p always tests/drivers/spi/spi_loopback/` `west build -b frdm_mcxn947//cpu0 -p always tests/drivers/spi/spi_loopback/ -DCONFIG_SPI_NXP_LPSPI_DMA=n` And found that all test cases pass for both DMA and CPU-based LPSPI drivers.
> 
> Can you confirm which pins you are looping for the test? It should be J2-8 and J2-10. And have you made any other modification to the board or build configurations. And what SHA you are testing on.

I think it has been fixed with #94786. The test failed on the previous commit (c7e6034) but is now successful for me as well.

csann, 2025-11-05:
My app now runs correctly with zephyr 4_3_0_rc2 and sdk 0.17.4 so I will close this issue.
Thanks to everyone who participated in fixing this bug. It feels good that I now run correctly with the latest build.

@decsny After comparing your code and my original code, I see where my code caused you to spend your time getting it to build. I now understand that if I need to submit a bug in the future, my code should be refined to eliminate all extraneous statements that do not directly pertain to the issue at hand. 

I also want to thank you for showing me how to use native control of CS. I was lead astray by others and made my code more complicated than it should have been.

decsny, 2025-11-06:
@csann no worries,  it's obvious you made an attempt at a useful bug report, and I have (very commonly) seen much worse ones with almost no info, and the reporter being less polite also.  so thanks for following up and embracing the mode of learning and collaborating.