zephyrproject-rtos/zephyr #94473: LVGL Widgets Demo stopped working in v4.1 on mimxrt1060_evk@A

laodzu, 2025-08-13:
### Describe the bug

The demo stopped working in v4.1 on my mimxrt1060_evk@A board.  In the new version I only get half a screen of random coloured pixels instead of the actual UI.

git bisecting yielded this commit as the culprit:

```
 commit 54472522477bccf3dd4a17d913c0116ee7e318d8
Author: Fabian Blatz <fabianblatz@gmail.com>
Date:   Sun Jan 26 18:53:26 2025 +0100

    west.yml: Sync LVGL module to 9.2.x
    
    Update the west yaml to point to the new LVGL version.
    
    Signed-off-by: Fabian Blatz <fabianblatz@gmail.com>

 west.yml | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```

This is already the second time that git bisect is not useful with LVGL problems because of this setup. 😭

This commit updates the LVGL repo from v9.2.0-346-g6decbb7f7 to v9.2.0-455-g1ed1ddd88. I skimmed the somewhat large diff but did not see an immediate culprit. Does anyone have an idea where I could start looking?

### Regression

- [x] This is a regression.

### Steps to reproduce

 west build -p -b mimxrt1060_evk@A/mimxrt1062/qspi --shield rk043fn02h_ct \
	 samples/modules/lvgl/demos -- -DCONFIG_LV_Z_DEMO_WIDGETS=y

### Relevant log output

```shell

```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

OS is Linux, Zephyr SDK 0.16.8

hakehuang, 2025-08-15:
> Does the display sample work?

the display sample works on this board on v4.2.0-1706-g12a866f7d3f.
```
west build -b mimxrt1060_evk@C/mimxrt1062/qspi -p always --shield=rk043fn02h_ct
```

laodzu, 2025-08-15:
Even though it is difficult to debug, I could make some progress here. In the meantime I believe we are - again - running out of memory somewhere and then as a result the whole program fails. I get to this conclusion by the following reasoning:

Looking into the working version, I realized that the display half full of random data is always there after running `mcux_elcdif_init`. Only in the good case this will be overwritten with the proper LVGL content. In the bad case this never happens and so the display ends up with this pattern. Potentially I think the display should be properly erased in the initialization phase, but currently this is not important.

The update of the display is triggered by the flush_thread in LVGL. I can put a break point on lvgl_display.c:31 to see this. In the good version this breakpoint is reached and on continuing the display refreshes normally. In the bad case, this breakpoint is never reached, and the system ends up in `k_sys_fatal_error_handler` as it crashed.

I encountered a memory related problem in this demo (#85306) before, so maybe this is similar again.

jhedberg, 2025-08-19:
Is this still reproducible with the latest main branch?

PR #96873 (2025-10-04): Fix LVGL crashes because of stack overflow
This fixes #94473
commit c60a870b (2025-10-06): samples: modules: lvgl: demos: Increase main stack to fix crash

The increased complexity of the LVGL code now overruns the 4 KiB stack
of the main thread. Increase it to 6 KiB to fix the error.

Fixes: #94473

(cherry picked from commit e05a2728176aaf71d36e8c7d9e5f832ccdf9fb74)