zephyrproject-rtos/zephyr #94499: Bluetooth: Controller: ISO: ISO PDUs not transmitted after first ~50 packets

ChangNice, 2025-08-14:
### Describe the bug

The bap_unicast_server (Peripheral) sends ISO SDUs to the bap_unicast_client (Central) continuously, but we found that only the first ~50 ISO SDUs can be received at the Central side. From the console log and btsnoop log, we confirmed that the ISO SDUs were sent to the Peripheral Controller. However, the sniffer (air packet) log shows there are no corresponding ISO PDUs.

### Regression

- [ ] This is a regression.

### Steps to reproduce

**Build/Flash Cmd:** 
west build -d .\build\hci_ipc\ -p always -b nrf5340_audio_dk/nrf5340/cpunet ./samples/bluetooth/hci_ipc -- -DOVERLAY_CONFIG='nrf5340_cpunet_iso-bt_ll_sw_split.conf'

west flash -d .\build\hci_ipc\ -r nrfjprog

west build -b nrf5340_audio_dk/nrf5340/cpuapp -d build/bap_unicast_server .\samples\bluetooth\bap_unicast_server\ -DEXTRA_CONF_FILE="overlay-all-bt_ll_sw_split.conf"

west flash -d .\build\bap_unicast_server\ -r nrfjprog

west build -b nrf5340_audio_dk/nrf5340/cpuapp -d build/bap_unicast_client .\samples\bluetooth\bap_unicast_client\ -DEXTRA_CONF_FILE="overlay-all-bt_ll_sw_split.conf"

west flash -d .\build\bap_unicast_client\ -r nrfjprog

**Reproduce:**

1. build/flash for bap_unicast_server and bap_unicast_client
2. demo will auto-scan and connect.
3. just wait bap_unicast_client console log stop, then you will see:
```
Incoming audio on stream 0x200051c8 len 9 (42)
Incoming audio on stream 0x200051c8 len 10 (43)
Incoming audio on stream 0x200051c8 len 11 (44)
Incoming audio on stream 0x200051c8 len 12 (45)

```

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Additional Context

**console logs:**
[bap_unicast_server_console_log.txt](https://github.com/user-attachments/files/21770087/bap_unicast_server_console_log.txt)
[bap_unicast_client_console_log.txt](https://github.com/user-attachments/files/21770088/bap_unicast_client_console_log.txt)

**btsnoop logs:**
[2025-08-14 161143_bap_unicast_server_btsnoop.log](https://github.com/user-attachments/files/21770098/2025-08-14.161143_bap_unicast_server_btsnoop.log)
[2025-08-14 161236_bap_unicast_client_btsnoop.log](https://github.com/user-attachments/files/21770101/2025-08-14.161236_bap_unicast_client_btsnoop.log)

**sniffer logs:**
[bap_unicast_server_ota.zip](https://github.com/user-attachments/files/21770127/bap_unicast_server_ota.zip)

Thalley, 2025-08-20:
Not 100% if this is a controller issue or not. There is a pending refactor of TX from the unicast server her: https://github.com/zephyrproject-rtos/zephyr/pull/90052 

Could you try that out and see if there's still a problem?

Thalley, 2025-08-21:
I am not able to reproduce this issue: 

Server log: 
```
Audio Stream 0x200083c8 started
Start: stream 0x20008374
Audio Stream 0x20008374 started
[00:00:09.846,282] <inf> lc3: extract_lc3_config: Extracting LC3 configuration values
[00:00:09.846,313] <inf> lc3: stream_lc3_init: Initializing sine wave data
[00:00:09.846,771] <inf> lc3: stream_lc3_init: Setting up LC3 encoder
[00:00:09.847,045] <inf> stream_tx: stream_tx_register: Registered 0x20008374 for TX
Incoming audio on stream 0x200083a0 len 40 (1000)
[00:00:19.798,950] <inf> stream_tx: tx_thread_func: Stream 0x20008374: Sent 1000 total SDUs of size 40
Incoming audio on stream 0x200083c8 len 40 (1000)
Incoming audio on stream 0x200083a0 len 40 (2000)
[00:00:29.800,964] <inf> stream_tx: tx_thread_func: Stream 0x20008374: Sent 2000 total SDUs of size 40
Incoming audio on stream 0x200083c8 len 40 (2000)
Incoming audio on stream 0x200083a0 len 40 (3000)
[00:00:39.798,858] <inf> stream_tx: tx_thread_func: Stream 0x20008374: Sent 3000 total SDUs of size 40
Incoming audio on stream 0x200083c8 len 40 (3000)
Incoming audio on stream 0x200083a0 len 40 (4000)
```

Client log:
```
Audio Stream 0x200073b8 connected
Streams connected
Starting streams
Audio Stream 0x200073b8 started
[00:00:18.120,910] <inf> lc3: extract_lc3_config: Extracting LC3 configuration values
[00:00:18.120,941] <inf> lc3: stream_lc3_init: Initializing sine wave data
[00:00:18.121,398] <inf> lc3: stream_lc3_init: Setting up LC3 encoder
[00:00:18.121,704] <inf> stream_tx: stream_tx_register: Registered 0x200073b8 for TX
Audio Stream 0x200073e0 started
Streams started
[00:00:27.713,562] <inf> stream_tx: tx_thread_func: Stream 0x20007390: Sent 1000 total SDUs of size 40
[00:00:28.145,690] <inf> stream_tx: tx_thread_func: Stream 0x200073b8: Sent 1000 total SDUs of size 40
Incoming audio on stream 0x200073e0 len 40 (1000)
[00:00:37.712,860] <inf> stream_tx: tx_thread_func: Stream 0x20007390: Sent 2000 total SDUs of size 40
[00:00:38.146,392] <inf> stream_tx: tx_thread_func: Stream 0x200073b8: Sent 2000 total SDUs of size 40
Incoming audio on stream 0x200073e0 len 40 (2000)
[00:00:47.712,860] <inf> stream_tx: tx_thread_func: Stream 0x20007390: Sent 3000 total SDUs of size 40
[00:00:48.145,690] <inf> stream_tx: tx_thread_func: Stream 0x200073b8: Sent 3000 total SDUs of size 40
Incoming audio on stream 0x200073e0 len 40 (3000)
[00:00:57.712,860] <inf> stream_tx: tx_thread_func: Stream 0x20007390: Sent 4000 total SDUs of size 40
[00:00:58.145,690] <inf> stream_tx: tx_thread_func: Stream 0x200073b8: Sent 4000 total SDUs of size 40
Incoming audio on stream 0x200073e0 len 40 (4000)
```

Building both samples as ` west build -b nrf5340_audio_dk/nrf5340/cpuapp --sysbuild`

When building the appcore, you should not specify `DEXTRA_CONF_FILE="overlay-all-bt_ll_sw_split.conf"` - That's is a config used for the network core.

ChangNice, 2025-08-22:
@cvinayak You are right, issue disappear, after add `-DCONFIG_BT_CTLR_ISOAL_SN_STRICT=n -DCONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y`

bap_unicast_server log:
Stream 0x20004a6c: Sent 1000 total SDUs of size 40
Incoming audio on stream 0x20004a9c len 40 (1000)
Incoming audio on stream 0x20004ac8 len 40 (1000)
Stream 0x20004a6c: Sent 2000 total SDUs of size 40
Incoming audio on stream 0x20004a9c len 40 (2000)
Incoming audio on stream 0x20004ac8 len 40 (2000)
Stream 0x20004a6c: Sent 3000 total SDUs of size 40

bap_unicast_client log:
Incoming audio on stream 0x200050e0 len 40 (1000)
[00:00:22.593,750] [0m<inf> stream_tx: Stream 0x20005090: Sent 1000 total SDUs of size 40[0m
[00:00:23.114,013] [0m<inf> stream_tx: Stream 0x200050b8: Sent 1000 total SDUs of size 40[0m
Incoming audio on stream 0x200050e0 len 40 (2000)
[00:00:35.094,451] [0m<inf> stream_tx: Stream 0x20005090: Sent 2000 total SDUs of size 40[0m
[00:00:35.614,013] [0m<inf> stream_tx: Stream 0x200050b8: Sent 2000 total SDUs of size 40[0m
Incoming audio on stream 0x200050e0 len 40 (3000)

ChangNice, 2025-08-22:
I didn't cherry-pick the commit of https://github.com/zephyrproject-rtos/zephyr/pull/90052.

I can try it at my side.

I am working on this code base:
```
PS C:\_DDM\zephyrproject\zephyr> git log
commit 347fb0a75a1fc637b6f7bf00e951ae57c89c2c47 (HEAD -> main, origin/main)
Author: Tomasz Bursztyka <tobu@bang-olufsen.dk>
Date:   Wed Aug 20 11:15:33 2025 +0200

    drivers: pwm: Use DEVICE_API relevantly on mspm0 driver

    It was missing thus it was not possible to get a valid answer
    from DEVICE_API_IS().

    Signed-off-by: Tomasz Bursztyka <tobu@bang-olufsen.dk>

commit 3b7a797d2dd792d7bad23c1aab4f4e203cc9b881
```