zephyrproject-rtos/zephyr #94517: drivers: can: mcux: flexcan does not switch the clock source based upon DTS

SynchronicIT, 2025-08-14:
### Describe the bug

The zephyr/drivers/can/can_mcux_flexcan.c driver can select the clock assignment and clock-src through DTS. 

FIlling both "clk-source = <4>" and "clocks = <&scg SCG_K4_SYSOSC_CLK 0xec>;" in the DTS will switch clock core freq. source but not the actual clock source. (thus the actual freq and number used to calculate the bitrate do not match. 
the clk-source value from the DTS is not used anywere, unless the flexcan peripheral has a clock-src register. The MCUX-W71 e.g. does not have this register and should be filled in the K4 clock peripheral. 

To reproduce, build one of the CAN bus examples for the frdm-w71 and try to communicate with a different can device. The bit rate will be more than 2% off, and therefore out of spec. 

I do not have the complete overview of which chipsets this same error occurs, but anyone not having the clock-src register in the the peripheral (CAN_CTRL1_CLKSRC_MASK).

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

henrikbrixandersen, 2025-08-15:
If a SoC does not have the clk-src register what do you expect settings the clk-source DTS property to do?

SynchronicIT, 2025-08-15:
The issue is not related to the clk-source property in principle. 
The actual clock assignment is not based on the "clocks" property and therefore could create a mismatch.
The clocks property should be leading and depending on the chipset, the correct register (CTRL1_CLKSRC or  CLOCK_SetIpSrc(kCLOCK_Can0, ...)

@henrikbrixandersen to answer your question: The clk-source property can be ignored when there is no clk-src register. But the clk-source property should actually not exist at all.

SynchronicIT, 2025-08-15:
As addition. The Zephyr driver relies on "FLEXCAN_Init(..) of the hal_nxp module for configuring the clock. 
Depending on the chipset, the hal driver takes a good job or a poor job. 
Perhaps this issue should be forwarded to the hal_nxp (and therefore the Mcuxpresso sdk of NXP.