zephyrproject-rtos/zephyr #94556: Bluetooth: Peripheral changes advertising address temporarily when connection fails

seankyer, 2025-08-15:
### Describe the bug

When the peripheral disconnects from the central, the peripheral device (running Zephyr host stack) will briefly advertise itself with a different HW address before returning to the original address. 

This occurs with the BT_PRIVACY option enabled; I expect the address to change periodically with this option enabled, yet this behavior seems to be a little glitchy and may require further investigation.

This issue is not blocking, as the device can just be ignored when it advertises with this random address.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Build and flash an example which uses CONFIG_BT_PRIVACY.
2. Observe the Zephyr device advertising with a private HW address
3. Connect with NRF Connect or ADI Attach
4. Have the Zephyr device close the connection
5. Monitor the BLE scanner and observe a new device appear momentarily before disappearing and the original HW address being advertised again.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- ADI Attach
- Zephyr v4.2.0

seankyer, 2025-08-16:
> > [@Thalley](https://github.com/Thalley) are you sure this is a host issue?
> 
> Oh, actually I'm not. Could be a controller issue indeed.
> 
> [@seankyer](https://github.com/seankyer) are you using the Zephyr controller, or something else?

We are not using the Zephyr controller; I asked internally and the consensus was that its most likely a host issue. With that being said, I'm afraid I haven't investigated it much further.

Thalley, 2025-08-17:
> We are not using the Zephyr controller; I asked internally and the consensus was that its most likely a host issue. With that being said, I'm afraid I haven't investigated it much further.

Are you able to generate and possibly share an HCI log? It should provide a good overview of whether the host is setting an incorrect address at some point, or if the controller is doing it. If possible, you can also try to use the Zephyr controller and see if you are seeing a similar issue

Thalley, 2025-08-19:
@seankyer Are you on Linux? If so, then there's a tool called `btmon` that is far better at showing an HCI trace: https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-tools.html#embedded-hci-tracing 

If you are to, can you try to make that work, and share the log of that? 

What I'm trying to determine is whether it is the Zephyr host, or your controller, that sets the unexpected address.

jhedberg, 2025-08-19:
FWIW, since there are no signs of misbehavior (at least relating to this bug) from `bt_hci_core` in the logs, I'd be inclined to think this is a controller issue (i.e. it autonomously doing some wonky stuff with the address). However, as @Thalley said, it'd be best to see the full HCI log using something like btmon.

seankyer, 2025-10-21:
Closing this since I am no longer able to reproduce.