zephyrproject-rtos/zephyr #94559: Versal NET RPU: west flash as documented errors out

yashi, 2025-08-15:
### Describe the bug

The [doc](https://docs.zephyrproject.org/latest/boards/amd/versalnet_rpu/doc/index.html#programming-and-debugging) for Versal NET RPU says:

> Build and flash in the usual way. Here is an example for the Hello World application.
> 
> ```
> # From the root of the zephyr repository
> west build -b versalnet_rpu samples/hello_world
> west flash
> ```
> 
> You should see the following message on the console:
> ```
> Hello World!
> ```

But it doesn't work. This is due to [this](https://github.com/zephyrproject-rtos/zephyr/pull/92464/commits/7fcc43d05ab2fe71ab4f36f134c64f61ee5c14b4) change:

### Regression

- [x] This is a regression.

### Steps to reproduce

1. vresion
2. update
3. build
4. flash

### Relevant log output

```shell
**Veresion**

git describe
v4.2.0-1732-gaa6f87dcbfee

**Update**

west update

**Build**

❯ rm -rf build && west build -p -b versalnet_rpu samples/hello_world       
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/yashi/src/zephyr-project/zephyr/samples/hello_world
-- CMake version: 3.31.6
-- Found Python3: /home/yashi/src/zephyr-project/.venv/bin/python (found suitable version "3.13.6", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/yashi/.cache/zephyr
-- Zephyr version: 4.2.99 (/home/yashi/src/zephyr-project/zephyr)
-- Found west (found suitable version "1.4.0", minimum required is "0.14.0")
-- Board: versalnet_rpu, qualifiers: amd_versalnet_rpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.2 (/home/yashi/zephyr-sdk-0.17.2)
-- Found toolchain: zephyr 0.17.2 (/home/yashi/zephyr-sdk-0.17.2)
-- Found Dtc: /opt/xilinx/Vitis/2024.2/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: /home/yashi/src/zephyr-project/zephyr/boards/amd/versalnet_rpu/versalnet_rpu.dts
-- Generated zephyr.dts: /home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/yashi/src/zephyr-project/zephyr/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/yashi/src/zephyr-project/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
/home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.dts:92.38-101.5: Warning (interrupt_provider): /soc/interrupt-controller@e2000000: Missing #address-cells in interrupt provider
Parsing /home/yashi/src/zephyr-project/zephyr/Kconfig
Loaded configuration '/home/yashi/src/zephyr-project/zephyr/boards/amd/versalnet_rpu/versalnet_rpu_defconfig'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/yashi/src/zephyr-project/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/yashi/src/zephyr-project/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/yashi/zephyr-sdk-0.17.2/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yashi/zephyr-sdk-0.17.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Found gen_kobject_list: /home/yashi/src/zephyr-project/zephyr/scripts/build/gen_kobject_list.py
-- Configuring done (3.5s)
-- Generating done (0.1s)
-- Build files have been written to: /home/yashi/src/zephyr-project/zephyr/build
-- west build: building application
[1/124] Preparing syscall dependency handling

[3/124] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.2.99 (/home/yashi/src/zephyr-project/zephyr), build: v4.2.0-1732-gaa6f87dcbfee
[124/124] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:          0 GB         0 GB
             RAM:       31040 B         2 GB      0.00%
             OCM:          0 GB         1 MB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.elf for board: versalnet_rpu

**Flash**

❯ west -v flash
-- west flash: rebuilding
cmake version 3.31.6 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake --build /home/yashi/src/zephyr-project/zephyr/build
ninja: no work to do.
-- west flash: using runner xsdb
runners.xsdb: xsdb /home/yashi/src/zephyr-project/zephyr/boards/amd/versalnet_rpu/support/xsdb.cfg /home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.elf
couldn't open "": no such file or directory
    while executing
"error "$msg""
    (procedure "::xsdb::device::program" line 210)
    invoked from within
"device program $pdi_file"
    (procedure "load_image" line 17)
    invoked from within
"load_image {*}$argv"
    (file "/home/yashi/src/zephyr-project/zephyr/boards/amd/versalnet_rpu/support/xsdb.cfg" line 40)
FATAL ERROR: command exited with status 1: xsdb /home/yashi/src/zephyr-project/zephyr/boards/amd/versalnet_rpu/support/xsdb.cfg /home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.elf
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: Linux (Debian Sid)
- Tool chain: Zephyr SDK
- Commit hash: v4.2.0-1732-gaa6f87dcbfee

yashi, 2025-08-15:
@kedareswararao, do you mind to take a look?

kedareswararao, 2025-08-16:
> [@kedareswararao](https://github.com/kedareswararao), do you mind to take a look?
Hi @yashi 
Earlier, the PDI was passed as an environment variable, but with the recent changes it has been moved to one of the runner arguments. I’ll send a pull request to update the documentation. Apologies for missing this update in the previous pull request.

Regards,
Kedar.

kedareswararao, 2025-10-18:
> [@kedareswararao](https://github.com/kedareswararao) it should be described in documentation. In qemu target xsdb is not used that's why PDI is not required. On the real board it is.

sure @michalsimek : will update the documentation and will send the pull request

closed 2025-12-19 by nashif