zephyrproject-rtos/zephyr #94561: stm32: i2s_stm32_sai driver does not release the last mem_block

puliceworx, 2025-08-15:
### Describe the bug

While implementing SAI, DRAIN, STOP, unable to refill men_slab completely causing a NOMEM error. 
Diagnosis showed that the last men_block is not freed in HAL_SAI_TxCpltCallback. This occurs on last_block check and no more data check. 

Only the previous block is free'd at the end of the callback. On last_block or no more blocks in the queue, free the streams men_block. 

Target: STM32U5xx 
Board: nucleo_u575zi_q
Diagnose: Modify sample\drivers\i2s\output to print free blocks after waiting a time after triggering the drain. 

### Regression

- [ ] This is a regression.

### Steps to reproduce

Modify the i2s\output sample to loop on free blocks to equal NUM_BLOCKS. This will loop forever. 

```
while(k_mem_slab_num_free_get(&tx_0_mem_slab) != NUM_BLOCKS) {
		
	printf("Waiting for all blocks to be free. %d of %d \n",
		k_mem_slab_num_free_get(&tx_0_mem_slab), NUM_BLOCKS);

	k_msleep(500);
}

```

west build -b nucleo_u575zi_q samples\drivers\i2s\output

### Relevant log output

```shell
*** Booting Zephyr OS build v4.2.0 ***
All I2S blocks written
Waiting for all blocks to be free. 16 of 20 
Waiting for all blocks to be free. 19 of 20 
Waiting for all blocks to be free. 19 of 20 
Waiting for all blocks to be free. 19 of 20 
Waiting for all blocks to be free. 19 of 20 
Waiting for all blocks to be free. 19 of 20 
Waiting for all blocks to be free. 19 of 20 
Waiting for all blocks to be free. 19 of 20
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

OS: Windows 11
Toolchain: Zephyr SDK 0.17.2
Version: v4.2.0

erwango, 2025-08-19:
@mariopaja Would you mind having a look ?

puliceworx, 2025-08-19:
> Hi [@puliceworx](https://github.com/puliceworx),
> 
> Is this issue only on DRAIN or also on STOP? DRAIN I could replicate and I have a solution already for it :)

This happens on STOP as well. 

When STOP is triggered last_block gets set. 

```
stream->last_block = true;
stream->state = I2S_STATE_STOPPING;

```

Then in HAL_SAI_TxCpltCallback 
```
if (stream->last_block) {
	LOG_DBG("TX Stopped ...");
	goto exit;
}

```

Does not free the last block and potentially leaves data in the queue. Unsure what should happen if there are more mem_blocks in the queue when stop is called.

mariopaja, 2025-08-19:
@puliceworx Yes I just noticed.

I already found the issue for both cases. An initial test that I did seems like it fixes both the issues.

I will do some more tests and open tomorrow a PR :)

STOP:

```
	ret = i2s_trigger(dev_i2s, I2S_DIR_TX, I2S_TRIGGER_STOP);
	if (ret < 0) {
		printf("Could not trigger I2S tx\n");
		return ret;
	}

	k_msleep(5000);
	
	ret = i2s_trigger(dev_i2s, I2S_DIR_TX, I2S_TRIGGER_START);
	if (ret < 0) {
		printf("Could not trigger I2S tx\n");
		return ret;
	}
```

LOG:

```
[00:00:00.001,000] <wrn> i2s_stm32_sai: Free mem slabs: 1
[00:00:00.002,000] <wrn> i2s_stm32_sai: Free mem slabs: 2
[00:00:00.004,000] <wrn> i2s_stm32_sai: Free mem slabs: 3
[00:00:00.005,000] <wrn> i2s_stm32_sai: Free mem slabs: 4
[00:00:00.007,000] <wrn> i2s_stm32_sai: Free mem slabs: 5
[00:00:00.008,000] <wrn> i2s_stm32_sai: Free mem slabs: 6
[00:00:00.010,000] <wrn> i2s_stm32_sai: Free mem slabs: 7
[00:00:00.011,000] <wrn> i2s_stm32_sai: Free mem slabs: 8
[00:00:00.013,000] <wrn> i2s_stm32_sai: Free mem slabs: 9
[00:00:00.014,000] <wrn> i2s_stm32_sai: Free mem slabs: 10
[00:00:00.015,000] <wrn> i2s_stm32_sai: Free mem slabs: 11
[00:00:00.017,000] <wrn> i2s_stm32_sai: Free mem slabs: 12
[00:00:00.018,000] <wrn> i2s_stm32_sai: Free mem slabs: 13
[00:00:00.020,000] <wrn> i2s_stm32_sai: Free mem slabs: 14
[00:00:00.021,000] <wrn> i2s_stm32_sai: Free mem slabs: 15
[00:00:00.023,000] <wrn> i2s_stm32_sai: Free mem slabs: 16
[00:00:05.023,000] <wrn> i2s_stm32_sai: Free mem slabs: 17
[00:00:05.024,000] <wrn> i2s_stm32_sai: Free mem slabs: 18
[00:00:05.025,000] <wrn> i2s_stm32_sai: Free mem slabs: 19
[00:00:05.027,000] <wrn> i2s_stm32_sai: Free mem slabs: 20
```

DRAIN:

```
	ret = i2s_trigger(dev_i2s, I2S_DIR_TX, I2S_TRIGGER_DRAIN);
	if (ret < 0) {
		printf("Could not trigger I2S tx\n");
		return ret;
	}
```

LOG:

```
[00:00:00.001,000] <wrn> i2s_stm32_sai: Free mem slabs: 1
[00:00:00.002,000] <wrn> i2s_stm32_sai: Free mem slabs: 2
[00:00:00.004,000] <wrn> i2s_stm32_sai: Free mem slabs: 3
[00:00:00.005,000] <wrn> i2s_stm32_sai: Free mem slabs: 4
[00:00:00.007,000] <wrn> i2s_stm32_sai: Free mem slabs: 5
[00:00:00.008,000] <wrn> i2s_stm32_sai: Free mem slabs: 6
[00:00:00.010,000] <wrn> i2s_stm32_sai: Free mem slabs: 7
[00:00:00.011,000] <wrn> i2s_stm32_sai: Free mem slabs: 8
[00:00:00.013,000] <wrn> i2s_stm32_sai: Free mem slabs: 9
[00:00:00.014,000] <wrn> i2s_stm32_sai: Free mem slabs: 10
[00:00:00.015,000] <wrn> i2s_stm32_sai: Free mem slabs: 11
[00:00:00.017,000] <wrn> i2s_stm32_sai: Free mem slabs: 12
[00:00:00.018,000] <wrn> i2s_stm32_sai: Free mem slabs: 13
[00:00:00.020,000] <wrn> i2s_stm32_sai: Free mem slabs: 14
[00:00:00.021,000] <wrn> i2s_stm32_sai: Free mem slabs: 15
[00:00:00.023,000] <wrn> i2s_stm32_sai: Free mem slabs: 16
[00:00:00.024,000] <wrn> i2s_stm32_sai: Free mem slabs: 17
[00:00:00.025,000] <wrn> i2s_stm32_sai: Free mem slabs: 18
[00:00:00.027,000] <wrn> i2s_stm32_sai: Free mem slabs: 19
[00:00:00.028,000] <wrn> i2s_stm32_sai: Free mem slabs: 20
```

puliceworx, 2025-08-19:
@mariopaja yes! That should cover all exits to free the block.

mariopaja, 2025-08-22:
@puliceworx @erwango i guess this issue can be closed

schweigstill, 2025-08-26:
@mariopaja It seems that the mem_block buffer patch with mem_block_tmp hasn't been merged to main branch. I spent the whole evening to figure out the resulting memory corruptions...

mariopaja, 2025-08-26:
@schweigstill sorry but I did not understand your issue issue. What do you mean it is not merged into main?

Do you have another issue related to mem_block?

schweigstill, 2025-08-26:
@mariopaja The problems were caused by a kind of "race condition". I started working on the buffer problem and made a similar change (with mem_block_tmp). When updating Zephyr this collided with your recent changes. I have to apologize for this confusion.

schweigstill, 2025-08-29:
@mariopaja Unfortunately I have to reopen this issue...

The problem is still existing. In HAL_SAI_TxCpltCallback() the wrong new buffer will be released with k_mem_slab_free() instead of the old buffer because it's pointer gets overwritten by stream->mem_block = item.buffer;. In your development branch this ist handled correctly by using a temporary pointer (mem_block_tmp) but on the main branch this change wasn't applied.

mariopaja, 2025-08-29:
@schweigstill I think you might be right. I think this happens during Stop. Can you provide more logs?

However, since this issue is not critical and Stop/Start & Drain work without raising any error. This issue would have low priority for me. 

I am a bit busy at the moment on other topics and a fix could last a bit longer.