zephyrproject-rtos/zephyr #94567: Cannot compile "hello world" for newly added lp_mspm0g3519 board

Eqqm4n, 2025-08-16:
### Describe the bug

Trying to compile the "hello world" example for the newly (uploaded as of yesterday at the time I write this) supported lp_mspm0g3519 board produces errors.  Searching for the terms `FACTORYVALUE` and `DL_FactoryRegion_initTrimTable` reveal them to be in files that are conditionally compiled.  It may be that the `DeviceFamily_PARENT` and `__MSPM0GX51X_TRIM_CACHE__` properties are not being set correctly.  However, even if I remove all the `#ifdef` code around these places I still get the errors, maybe due to some other conditionals preventing these entire files from being part of the build.

### Regression

- [ ] This is a regression.

### Steps to reproduce

 west build -b lp_mspm0g3519 ./hello_world/ --build-dir ./hello_world/build/ -p

### Relevant log output

```shell
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\nVent_work\zephyr_root\zephyr_repo\zephyr-sdk-0.17.0\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe  -gdwarf-4 -Os zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map,C:/nVent_work/zephyr_root/zephyr_repo/nvent_projects/hello_world/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/ti/mspm0/lib..__modules__hal__ti__mspm0.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -LC:/nVent_work/zephyr_root/zephyr_repo/nvent_projects/hello_world/build/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m0plus  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF -Lc:/nvent_work/zephyr_root/zephyr_repo/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v6-m/nofp -lc -lgcc && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\nVent_work\zephyr_root\zephyr_repo\nvent_projects\hello_world\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
c:/nvent_work/zephyr_root/zephyr_repo/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: modules/ti/mspm0/lib..__modules__hal__ti__mspm0.a(dl_sysctl_mspm0gx51x.c.obj): in function `DL_SYSCTL_configSYSPLL':
C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/modules/hal/ti/mspm0/source/ti/driverlib/m0p/sysctl/dl_sysctl_mspm0gx51x.c:58: undefined reference to `DL_FactoryRegion_initTrimTable'
c:/nvent_work/zephyr_root/zephyr_repo/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/modules/hal/ti/mspm0/source/ti/driverlib/m0p/sysctl/dl_sysctl_mspm0gx51x.c:120: undefined reference to `FACTORYVALUE'
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

Windows 11 Zephyr SDK 0.17.0

dsseng, 2025-08-16:
This seems to be a Windows platform-specific issue: `west build -b lp_mspm0g3519 samples/hello_world/ -p` works fine for me on Linux.

Eqqm4n, 2025-08-16:
Tried again doing it right out of the `samples` directory to the same result:
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\nVent_work\zephyr_root\zephyr_repo\zephyr-sdk-0.17.0\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe  -gdwarf-4 -Os zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map,C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/ti/mspm0/lib..__modules__hal__ti__mspm0.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -LC:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/zephyr/build/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m0plus  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF -Lc:/nvent_work/zephyr_root/zephyr_repo/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v6-m/nofp -lc -lgcc && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\nVent_work\zephyr_root\zephyr_repo\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
c:/nvent_work/zephyr_root/zephyr_repo/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: modules/ti/mspm0/lib..__modules__hal__ti__mspm0.a(dl_sysctl_mspm0gx51x.c.obj): in function `DL_SYSCTL_configSYSPLL':
C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/modules/hal/ti/mspm0/source/ti/driverlib/m0p/sysctl/dl_sysctl_mspm0gx51x.c:58: undefined reference to `DL_FactoryRegion_initTrimTable'
c:/nvent_work/zephyr_root/zephyr_repo/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/modules/hal/ti/mspm0/source/ti/driverlib/m0p/sysctl/dl_sysctl_mspm0gx51x.c:120: undefined reference to `FACTORYVALUE'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\nVent_work\zephyr_root\zephyr_repo\zephyrproject\zephyr\build'

```
It is also curious as to where these things are even supposed to be defined, because doing searches only reveals places where they are checked, but not set:
```
PS C:\nVent_work\zephyr_root\zephyr_repo\zephyrproject> Get-ChildItem -Path .\*.c -Recurse | Select-String -Pattern "DeviceFamily_PARENT"

modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0c110x.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0C110X)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0c110x.c:169:#endif /* DeviceFamily_PARENT_MSPM0C110X */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0g1x0x_g3x0x.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0G1X0X_G3X0X)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0g1x0x_g3x0x.c:320:#endif /* DeviceFamily_PARENT_MSPM0G1X0X_G3X0X */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0gx51x.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0GX51X)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0gx51x.c:382:#endif /* DeviceFamily_PARENT_MSPM0GX51X */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0h321x.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0H321X)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0h321x.c:240:#endif /* DeviceFamily_PARENT_MSPM0H321X */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0l111x.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0L111X)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0l111x.c:223:#endif /* DeviceFamily_PARENT_MSPM0L111X */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0l11xx_l13xx.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0L11XX_L13XX)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0l11xx_l13xx.c:146:#endif /* DeviceFamily_PARENT_MSPM0L11XX_L13XX */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0l122x_l222x.c:36:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0L122X_L222X)
modules\hal\ti\mspm0\source\ti\driverlib\m0p\sysctl\dl_sysctl_mspm0l122x_l222x.c:269:#endif /* DeviceFamily_PARENT_MSPM0L122X_L222X */
modules\hal\ti\mspm0\source\ti\driverlib\m0p\dl_factoryregion.c:37:#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0GX51X)
[... cut ...]
modules\hal\ti\simplelink_lpf3\source\ti\drivers\utils\Random.c:117:#elif (DeviceFamily_PARENT == DeviceFamily_PARENT_CC23X0)
```
Note that I did that search because the missing FACTORYVALUE is a symbol that only gets compiled if `#if (DeviceFamily_PARENT == DeviceFamily_PARENT_MSPM0GX51X)` evaluates as true.  But as stated I cannot find any place in the Zephyr code that sets `DeviceFamily_PARENT`.  Presumably that means it either never gets defined or the source file for FACTORYVALUE isn't compiled.

Eqqm4n, 2025-08-16:
Alright if I modify the `samples/hello_world/CMakeLists.txt` to force the inclusion of the file holding FACTORYVALUE then I get a clean compile, so there's at least a hack to address this issue.
```
target_sources(app PRIVATE src/main.c C:/nVent_work/zephyr_root/zephyr_repo/zephyrproject/modules/hal/ti/mspm0/source/ti/driverlib/m0p/dl_factoryregion.c)
```

ssekar15, 2025-08-19:
I'm able to reproduce, will work on to fix it.

Eqqm4n, 2025-08-27:
@ssekar15 As a side note, even if this fixed the issue, it would still be a problem for me.  My group needs to get CANopenNode 4.0 running on the TI MSPM0G3519, which means I have to pull code from one fork that runs CANopenNode 4.0, then a different fork from TI's internal development site that includes support for the on-chip CAN peripheral for the IC.