zephyrproject-rtos/zephyr #94637: native_sim: Implement RTC-api for low-level rtc driver

clamattia, 2025-08-19:
### Summary

- native sim has a "low-level" rtc driver
- it was created before the RTC-Interface

### Describe the solution you'd like

- Implement the RTC-driver-API for the native_sim RTC-driver

### Alternatives

TBD

### Additional Context

TBD

bjarki-andreasen, 2025-08-20:
> > The native_sim boards actually do have an emulated RTC which behaves like real hardware:
> 
> I was not aware. Maybe the [docs ](https://docs.zephyrproject.org/latest/boards/native/native_sim/doc/index.html) need updating? They say:
> 
> > Note this device does not yet have an [RTC API compatible driver](https://docs.zephyrproject.org/latest/hardware/peripherals/rtc.html#rtc-api).

That needs updating :)

> > this one is not persistent though ....
> 
> Do you mean across restarts of the native_sim app?

yes

clamattia, 2025-08-20:
> That needs updating :)

Closing this in favor of: https://github.com/zephyrproject-rtos/zephyr/issues/94742

closed 2026-09-08 by github-actions[bot]