zephyrproject-rtos/zephyr #94667: kernel: mem_heap: k_heap_api: kernel.k_heap_api fails

katgiadla, 2025-08-19:
### Describe the bug

Test `tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api` fails.

Observed for `nrf54l15dk/nrf54l15/cpuflpr`.

### Regression

- [x] This is a regression.

### Steps to reproduce

1. have `nrf54l15dk/nrf54l15` connected
2. go to your zephyr dir
3. call `./scripts/twister -s kernel.k_heap_api -p nrf54l15dk/nrf54l15/cpuflpr --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

### Relevant log output

```shell
*** Booting Zephyr OS build v4.2.0-1744-g1cb0b6a7ffa5 ***
Running TESTSUITE k_heap_api
===================================================================
START - test_k_heap_aligned_alloc
 FAIL - test_k_heap_aligned_alloc in 0.001 seconds
===================================================================
...
===================================================================
START - test_z_k_heap_double_free
 FAIL - test_z_k_heap_double_free in 0.001 seconds
===================================================================
...
SUITE FAIL -  86.67% [k_heap_api]: pass = 13, fail = 2, skip = 0, total = 15 duration = 0.225 seconds
 - FAIL - [k_heap_api.test_k_heap_aligned_alloc] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_alloc] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_alloc_fail] duration = 0.003 seconds
 - PASS - [k_heap_api.test_k_heap_alloc_pending] duration = 0.006 seconds
 - PASS - [k_heap_api.test_k_heap_alloc_pending_null] duration = 0.201 seconds
 - PASS - [k_heap_api.test_k_heap_array_get] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_calloc] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_free] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_min_size] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_realloc] duration = 0.002 seconds
 - PASS - [k_heap_api.test_k_heap_realloc_fail] duration = 0.003 seconds
 - PASS - [k_heap_api.test_k_heap_realloc_null] duration = 0.001 seconds
 - PASS - [k_heap_api.test_k_heap_realloc_zero] duration = 0.001 seconds
 - PASS - [k_heap_api.test_kheap_alloc_in_isr_nowait] duration = 0.001 seconds
 - FAIL - [k_heap_api.test_z_k_heap_double_free] duration = 0.001 seconds
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.17.0
- Commit SHA or Version used: 1cb0b6a7ffa5f76daa873ec4345e8fe35380c5b3

### Additional Context

Bisect indicates for 7f57759f9d7c8a1371df1d2f8cdd61a3461d06e2

dsseng, 2025-08-19:
`west build -p -b native_sim/native/64 tests/kernel/mem_heap/k_heap_api -T kernel.k_heap_api -DCONFIG_ASAN=y -t run` reproduces the issue, so it's not nRF platform-specific

nordic-piks, 2025-08-28:
@JarmouniA there is no more logs from this failing test case.
I checked it and found that the test expects assert to happen but for this platform L15/cpuflpr they are disabled by default.
Probably that test should explicit add CONFIG_ASSERT=y in prj.conf.
However when I did it tests fails but differently, but for known reason, related to platform limitation not the test itself.
So this issue can be ignored for L15/cpuflpr.