zephyrproject-rtos/zephyr #94681: Add protected storage (PS) settings backend

seankyer, 2025-08-19:
### Is your feature request related to a problem? Please describe.

We have an TFM ITS backend for the Settings subsystem https://github.com/zephyrproject-rtos/zephyr/commit/cba074230d874eb0f7a01b8ebe96d95c11e10c8a we should also create a TFM PS backend.
https://github.com/zephyrproject-rtos/zephyr/pull/87778

### Describe the solution you'd like

Extend the effort of providing secure persistent storage of settings in Zephyr by creating a settings_ps backend allowing the settings subsystem running on a non-secure target running with TFM to access disk.

tomi-font, 2025-08-20:
I don't quite agree. Why would we want that? And why not for instance just change the TF-M ITS backend to use the PS instead?
The PS is designed to store user data and to be less constrained in size compared to the ITS which is mainly for sensitive secrets.
It would make more sense to have the Settings backend store to the PS only and not the ITS.

seankyer, 2025-08-22:
> > Oh, I thought during the security WG review (a while ago now) we agreed there would be backends for ITS and PS? More options for the user essentially.
> 
> I'm a contributor and haven't been to that meeting, is there some transcript, video or a brief of it? Thanks

Not sure. @ceolin ran the meeting. I don't know if it was recorded or not? I actually don't even remember which week it was anymore...

dsseng, 2025-08-25:
> > SETTINGS_PSA?
> 
> Hmm yeah why not, though I would keep the `TFM` prefix as the backend is only meant to be used with TF-M.
> 
> > Both options would use PSA APIs (be it SECURE_STORAGE or a TF-M backed API)
> 
> I don't think we want to allow the backend to be used with the Secure storage subsystem. And besides, the API indeed is the same in any case, and the PSA APIs are not just TF-M-backed. 🙂 (TF-M just happens to be one implementation of them.)

Well, it currently works with secure storage. Are there better options to store settings in an encrypted form (and maybe with integrity) without TF-M?

tomi-font, 2025-08-27:
> Well, it currently works with secure storage. Are there better options to store settings in an encrypted form (and maybe with integrity) without TF-M?

Absolutely; using the PSA Secure Storage API directly.
This Settings backend has quite some drawbacks in its current implementation, that's part of the reason why I want to keep it for TF-M only. The original intent was just to allow storing data to NVM for those boards that wouldn't otherwise be able to when using TF-M.

tomi-font, 2025-08-27:
> Well, as described in the original issue, e.g. for Bluetooth settings the app would typically require a Settings implementation.

And it will get that from any of the many available options e.g. ZMS, NVS... If TF-M is not in use there will be no problem with the NVM not being available to the application,

closed 2025-12-10 by fabiobaltieri