zephyrproject-rtos/zephyr #94727: NXP PXP in LVGL rendering issues

pdgendt, 2025-08-20:
### Describe the bug

When enabling the PXP in LVGL (not DMA PXP in Zephyr's ELCDIF display driver), the output has many artifacts. Sometimes completely garbled, or a shifted image.
See video.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Build for the i.MX RT1064 Dev kit with the `rk043fn02h_ct` shield connected. Make sure buffer alignments are set, otherwise tasks aren't handed of to the PXP to begin with.

```shell
$ west build -p -b mimxrt1064_evk --shield rk043fn02h_ct samples/modules/lvgl/demos -- -DCONFIG_LV_USE_PXP=y -DCONFIG_LV_DRAW_BUF_ALIGN=32 -DCONFIG_LV_ATTRIBUTE_MEM_ALIGN_SIZE=32 -DCONFIG_MAIN_STACK_SIZE=16384
```

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: Ubuntu 22.04
- Toolchain: Zephyr SDK 0.17.2
- `main` at f1b4c7c9929ca93f5fb7b37bbc35a80c1134f4f7

### Additional Context

https://github.com/user-attachments/assets/607216f2-00a7-4c07-bbf7-cacf3ff3107b

faxe1008, 2025-08-20:
Possibly same issue @Finomnis :
https://github.com/zephyrproject-rtos/zephyr/issues/94185

faxe1008, 2025-08-31:
@Finomnis @pdgendt maybe a combination of these patches solves your issues

https://github.com/lvgl/lvgl/pull/8764

https://github.com/zephyrproject-rtos/zephyr/pull/94744