zephyrproject-rtos/zephyr #94794: lorawan: nvm: fuota: Mic error

J0hann3, 2025-08-21:
### Describe the bug

I'm working on the sample: `samples\subsys\lorawan\fuota` to update the firmware on a samd21, but I've got the error "Mic failed" multiple times even though everything should be fine.

I found out that, when using the configs `CONFIG_LORAWAN_NVM_SETTINGS`, `CONFIG_SETTINGS` and `CONFIG_NVS`, the structure `LoRaMacNvmData_t` would be saved in non-volatile memory. But the problem is that some variables are pointers. For example, `Nvm.MacGroup2.MulticastChannelList[0].DownLinkCounter` is a pointer, so during the initialization of the LoRa `lorawan_start()` it will restore the data stored in NVM. However, if the binary has changed in the meantime, the pointer will not anymore point to the right value. And when trying to start a multicast session, it will reject all the fragments according to this random value in memory with the error `Mic failed`.

The only easy fix that I found is to not use `CONFIG_NVS` and store the devnonce in the eeprom myself.

### Regression

- [ ] This is a regression.

### Steps to reproduce

I'm using the sample `samples\subsys\lorawan\fuota` as a base, which I have modified slightly to work on a samd21.
I added the following to the device tree:
```
&sercom4 {
	status = "okay";
	compatible = "atmel,sam0-spi";
	dipo = <3>;
	dopo = <0>;
	#address-cells = <1>;
	#size-cells = <0>;

	pinctrl-0 = <&sercom4_spi_default>;
	pinctrl-names = "default";
	cs-gpios = <&portb 10 GPIO_ACTIVE_LOW>, <&portb 5 GPIO_ACTIVE_LOW>;

	lora0: sx1262@0 {
		compatible = "semtech,sx1262";
    	reg = <0>;
		status = "okay";

        spi-max-frequency = <1000000>;
		label = "SX1262";
        reset-gpios = <&porta 21 GPIO_ACTIVE_LOW>;
        busy-gpios = <&porta 20 GPIO_ACTIVE_HIGH>;
        dio1-gpios = <&porta 15 GPIO_ACTIVE_HIGH>;
		dio2-tx-enable;
	};

	mx25r80: mx25r8035f@1 {
		status = "okay";
		compatible = "jedec,spi-nor";
		reg = < 1 >;
		wp-gpios = <&porta 24 GPIO_ACTIVE_LOW>;
		spi-max-frequency = < 1000000 >;
		has-dpd;
		dpd-wakeup-sequence = <30000 20 35000>;
		t-enter-dpd = <10000>;
		t-exit-dpd = <35000>;
		mxicy,mx25r-power-mode = "low-power";
		jedec-id = [ C2 28 14 ];
		size = <DT_SIZE_M(8)>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = < 0x1 >;
			#size-cells = < 0x1 >;
			slot1_partition: partition@0 {
				label = "image-1";
				reg = < 0x0 DT_SIZE_K(220) >;
			};
			scratch_partition: partition@37000 {
				label = "image-scratch";
				reg = < 0x37000 DT_SIZE_K(64) >;
			};
			storage_partition: partition@48000 {
			label = "storage";
			reg = <0x48000 DT_SIZE_K(16)>;
			};
		};
	};
};

&flash0 {
	erase-block-size = < 0x100 >;
	partitions {
		compatible = "fixed-partitions";
		#address-cells = < 0x1 >;
		#size-cells = < 0x1 >;
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = < 0x0 DT_SIZE_K(32) >;
			read-only;
		};
		slot0_partition: partition@8000 {
			label = "image-0";
			reg = < 0x8000 DT_SIZE_K(220) >;
		};
	};
};
```
I aslo added this to the sysbuild.conf:
```
SB_CONFIG_MCUBOOT_MODE_SWAP_SCRATCH=y
SB_CONFIG_BOOT_SIGNATURE_TYPE_NONE=y
```
and this for the mcuboot.conf:
```
CONFIG_MCUBOOT_LOG_LEVEL_DBG=y
CONFIG_MCUBOOT_UTIL_LOG_LEVEL_DBG=y

CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
CONFIG_MAIN_STACK_SIZE=4096
```
as well as this for the prj.conf:
```
CONFIG_LORAWAN_FRAG_TRANSPORT_MIN_FRAG_SIZE=80
CONFIG_LORAWAN_FRAG_TRANSPORT_MAX_FRAG_SIZE=200
CONFIG_LORAWAN_FRAG_TRANSPORT_MAX_REDUNDANCY=1
CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
CONFIG_LORAWAN_FRAG_TRANSPORT_DECODER_LOWMEM=y
```

### Relevant log output

```shell
[00:02:00.837,000] <dbg> lorawan_services: lorawan_services_class_c_start: Switched to class C
[00:02:03.628,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:02:03.629,000] <err> lorawan: McpsIndication failed : MIC fail
[00:02:03.653,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving LoRaWAN settings
[00:02:03.654,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Crypto version: 16777472, DevNonce: 12, JoinNonce: 11522814
[00:02:03.654,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup2
[00:02:04.788,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:02:04.789,000] <err> lorawan: McpsIndication failed : MIC fail
[00:02:06.042,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:02:06.042,000] <err> lorawan: McpsIndication failed : MIC fail
[00:02:07.201,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:02:07.202,000] <err> lorawan: McpsIndication failed : MIC fail
[00:02:08.455,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:02:08.455,000] <err> lorawan: McpsIndication failed : MIC fail
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

shiwzhi, 2025-09-04:
I was having the same problem. After some experiments, I found out that I needed to set the Gen App Key and Application Key to the same in ChirpStack.

JordanYates, 2025-12-12:
> I have been running into this too. The fix is just to wait for the new backend?

There are a few different options:

1. As mentioned by @J0hann3, there are workarounds available by disabling the builtin storage and saving the devnonce manually.
2. Update https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/lorawan/nvm/lorawan_nvm_settings.c to ensure it doesn't save any structs with pointers.
3. Add a new backend in the same location that only stores and restores the devnonce
4. Wait for someone to implement LoRaWAN over the `lora-basics-modem` library

I would be happy to review any PR implementing 2, 3 or 4. At the moment I am two steps removed from this problem, as I'm not actively using LoRaWAN, and I've never used the settings subsystem.

edit: It looks like there is more hope on number 4 than I thought, it looks like @carlocaione is working on it (from #100974 description)