zephyrproject-rtos/zephyr #94848: build break for samples/basic/minimal for board s32k148_evb

keykur111, 2025-08-22:
### Describe the bug

I am trying to build https://docs.zephyrproject.org/latest/samples/basic/minimal/README.html
especially Multithreading enabled, no preemption, timers disabled configuration.

i am getting below error
`01_zephyrproject/zephyr/soc/nxp/s32/s32k1/./soc.h:10:10: fatal error: fsl_port.h: No such file or directory
   10 | #include <fsl_port.h>`
getting this build error

### Regression

- [ ] This is a regression.

### Steps to reproduce

west build -b s32k148_evb -d build/s32k148_evb/mt-no-preempt-no-timers samples/basic/minimal -DCONF_FILE="common.conf mt.conf no-preempt.conf no-timers.conf arm.conf" 

### Relevant log output

```shell
-- west build: making build dir /localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /localpath/zephyrproject/zephyr/samples/basic/minimal
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/localpath/.cache/zephyr
-- Zephyr version: 4.2.99 (/localpath/zephyrproject/zephyr)
-- Found west (found suitable version "1.4.0", minimum required is "0.14.0")
-- Board: s32k148_evb, qualifiers: s32k148
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.2 (/home/localpath/zephyr-sdk-0.17.2)
-- Found toolchain: zephyr 0.17.2 (/home/localpath/zephyr-sdk-0.17.2)
-- Found Dtc: /home/localpath/zephyr-sdk-0.17.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6")
-- Found BOARD.dts: /localpath/zephyrproject/zephyr/boards/nxp/s32k148_evb/s32k148_evb.dts
-- Generated zephyr.dts: /localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers/zephyr/zephyr.dts
-- Generated pickled edt: /localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers/zephyr/edt.pickle
-- Generated devicetree_generated.h: /localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers/zephyr/include/generated/zephyr/devicetree_generated.h
Parsing /localpath/zephyrproject/zephyr/samples/basic/minimal/Kconfig
Loaded configuration '/localpath/zephyrproject/zephyr/boards/nxp/s32k148_evb/s32k148_evb_defconfig'
Merged configuration '/localpath/zephyrproject/zephyr/samples/basic/minimal/common.conf'
Merged configuration '/localpath/zephyrproject/zephyr/samples/basic/minimal/mt.conf'
Merged configuration '/localpath/zephyrproject/zephyr/samples/basic/minimal/no-preempt.conf'
Merged configuration '/localpath/zephyrproject/zephyr/samples/basic/minimal/no-timers.conf'
Merged configuration '/localpath/zephyrproject/zephyr/samples/basic/minimal/arm.conf'
Configuration saved to '/localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers/zephyr/.config'
Kconfig header saved to '/localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/localpath/zephyr-sdk-0.17.2/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/localpath/zephyr-sdk-0.17.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Found gen_kobject_list: /localpath/zephyrproject/zephyr/scripts/build/gen_kobject_list.py
-- Configuring done
-- Generating done
-- Build files have been written to: /localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers
-- west build: building application
[1/134] Preparing syscall dependency handling

[2/134] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.2.99 (/localpath/zephyrproject/zephyr), build: v4.2.0-1067-ga8ecd9404454
[9/134] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
ccache /home/localpath/zephyr-sdk-0.17.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCPU_S32K148 -DFSL_SDK_DISABLE_DRIVER_CLOCK_CONTROL=1 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGc
In file included from /localpath/zephyrproject/zephyr/modules/cmsis_6/./cmsis_core_m.h:24,
                 from /localpath/zephyrproject/zephyr/modules/cmsis_6/./cmsis_core.h:10,
                 from /localpath/zephyrproject/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24,
                 from /localpath/zephyrproject/zephyr/include/zephyr/arch/arm/asm_inline.h:18,
                 from /localpath/zephyrproject/zephyr/include/zephyr/arch/arm/arch.h:30,
                 from /localpath/zephyrproject/zephyr/include/zephyr/arch/cpu.h:19,
                 from /localpath/zephyrproject/zephyr/include/zephyr/kernel_includes.h:36,
                 from /localpath/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from /localpath/zephyrproject/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /localpath/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:9:
/localpath/zephyrproject/zephyr/soc/nxp/s32/s32k1/./soc.h:10:10: fatal error: fsl_port.h: No such file or directory
   10 | #include <fsl_port.h>
      |          ^~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /localpath/zephyrproject/zephyr/build/s32k148_evb/mt-no-preempt-no-timers
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

linux

keykur111, 2025-08-22:
Upon debugging i came to know that `CONFIG_SERIAL=n ` in [common.conf](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/basic/minimal/common.conf) is creating issue.

keykur111, 2025-08-26:
Thanks @decsny it looks fine on latest zephyr version

keykur111, 2025-08-28:
> [@keykur111](https://github.com/keykur111) can you check if it worked on 4.2.0 ?
@decsny Yes same issue with 4.2.0

keykur111, 2025-08-28:
> Hi, I will look at it next week. Thanks

thank you 
### I have tried below things as **workaround**

**When i build with CONFIG_SERIAL=y.** 
in ninja.dep
'/home/kpit/keyur/06_kpi/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/port/fsl_port.h  ˆÿÿÿh   /home/kpit/keyur/06_kpi/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/common/fsl_common.h  ‡ÿÿÿL   '

**so i included fsl_port.h with CONFIG_SERIAL=n using Below CMAKE**
`set(PUF_path $ENV{ZEPHYR_BASE}/../modules/hal/nxp/mcux/mcux-sdk-ng/drivers/port)
zephyr_sources(${PUF_path}//fsl_port.h)  
zephyr_include_directories(${PUF_path})`
help from https://github.com/zephyrproject-rtos/zephyr/discussions/71571

my build works but i am not sure about the minimalism i have achieved or not

manuargue, 2025-09-07:
please check if #95622 solves your issue @keykur111 , thanks