zephyrproject-rtos/zephyr #94906: UART async_api sample crashes when built for  xiao_ble board

sidcha, 2025-08-24:
### Describe the bug

I'm trying to run the uart async API sample on xiao_ble board and see a crash after a call to uart_rx_enable(). I went trough the documentation and the sample's readme, but nothing indicates a direct incompatibility.

### Regression

- [ ] This is a regression.

### Steps to reproduce

west build -p always -b xiao_ble samples/drivers/uart/async_api/ &&
west flash &&
west rtt

### Relevant log output

```shell
[00:00:00.287,841] <inf> udc_nrf: Preinit
*** Booting Zephyr OS build v4.2.0-2289-gbd8a2d3db4ee ***
[00:00:00.288,085] <inf> udc_nrf: Initialized
[00:00:00.292,297] <inf> udc_nrf: SUSPEND state detected
[00:00:00.393,615] <inf> udc_nrf: Reset
[00:00:00.393,646] <inf> udc_nrf: RESUMING from suspend
[00:00:00.393,676] <inf> udc: ep 0x80 is not halted|disabled
[00:00:00.508,026] <inf> udc_nrf: Reset
[00:00:00.508,056] <inf> udc: ep 0x80 is not halted|disabled
[00:00:00.635,467] <inf> udc: Sequence 2 not completed
[00:00:00.637,451] <inf> udc: Sequence 2 not completed
[00:00:00.639,465] <inf> udc: Sequence 2 not completed
[00:00:05.288,299] <inf> sample: Loop 0: Sending 4 packets
[00:00:05.288,330] <err> os: ***** USAGE FAULT *****
[00:00:05.288,360] <err> os:   Illegal use of the EPSR
[00:00:05.288,360] <err> os: r0/a1:  0x0000c728  r1/a2:  0x20003380  r2/a3:  0x00000013
[00:00:05.288,391] <err> os: r3/a4:  0xffffffff r12/ip:  0x00000000 r14/lr:  0x000002ad
[00:00:05.288,391] <err> os:  xpsr:  0x00000000
[00:00:05.288,391] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:05.288,421] <err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0
[00:00:05.288,452] <err> os: Current thread: 0x20001060 (unknown)
[00:00:05.540,618] <err> os: Halting system
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: Linux
- Toolchain: Zephyr SDK
- Commit: bd8a2d3db4ee39eaea5c8ae84295bcbeead5dfb9

### Additional Context

I have the following local changes to prj.conf to flash and view rtt logs with jlink.

```diff
diff --git a/samples/drivers/uart/async_api/prj.conf b/samples/drivers/uart/async_api/prj.conf
index ce46f814605..7a84791a491 100644
--- a/samples/drivers/uart/async_api/prj.conf
+++ b/samples/drivers/uart/async_api/prj.conf
@@ -1,3 +1,13 @@
+CONFIG_LOG=y
+CONFIG_PRINTK=y
+CONFIG_UART_CONSOLE=n
+CONFIG_RTT_CONSOLE=y
+CONFIG_USE_SEGGER_RTT=y
+
+# Override configs from xiao_ble defconfig.
+CONFIG_BUILD_OUTPUT_UF2=n
+CONFIG_USE_DT_CODE_PARTITION=n
+
 CONFIG_LOG=y
 CONFIG_NET_BUF=y
 CONFIG_SERIAL=y
```

dcpleung, 2025-08-25:
This is unique to the nRF UARTE driver, where you have to specify `CONFIG_UART_*_ASYNC=y` to enable async operations. My recommendation is to add a board config overlay in the sample so this (and other necessary) configs are added when building the sample.

sidcha, 2025-08-27:
I see. Thanks for clarifying. I'll add it to my custom board config. Closing as resolved.