zephyrproject-rtos/zephyr #94916: STM32 USB Regression

ExtremeGTX, 2025-08-24:
### Describe the bug

After updating to latest main zephyrproject-rtos/zephyr@617b71bc174be034b6b910792ea922c47b568068 USB is not working any more on STM32F769i_disco

### Regression

- [x] This is a regression.

### Steps to reproduce

1. `west build -b stm32f769i_disco ./samples/subsys/usb/cdc_acm/ `
2.  `copy ./build/zephyr/zephyr.bin <STLink MSC Partition>`

### Relevant log output

```shell
*** Booting Zephyr OS build v4.2.0-2290-g617b71bc174b ***
[00:00:00.010,000] <inf> cdc_acm_echo: USB device support enabled
[00:00:00.010,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.016,000] <inf> cdc_acm_echo: USBD message: Device suspended
[00:00:00.130,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:00.167,000] <err> usbd_core: unrecoverable error -22, ep 0x00, buf 0x20023bf8
[00:00:00.167,000] <inf> cdc_acm_echo: USBD message: Stack error
[00:00:05.689,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:05.795,000] <err> usbd_core: unrecoverable error -22, ep 0x00, buf 0x20023c1c
[00:00:05.795,000] <inf> cdc_acm_echo: USBD message: Stack error
[00:00:11.316,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:11.419,000] <err> usbd_core: unrecoverable error -22, ep 0x00, buf 0x20023bf8
[00:00:11.419,000] <inf> cdc_acm_echo: USBD message: Stack error
[00:00:16.936,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:17.039,000] <err> usbd_core: unrecoverable error -22, ep 0x00, buf 0x20023c1c
[00:00:17.039,000] <inf> cdc_acm_echo: USBD message: Stack error
[00:00:22.046,000] <inf> cdc_acm_echo: USBD message: Device suspended
```

### Impact

Showstopper – Prevents release or major functionality; system unusable.

### Environment

- OS:  Ubuntu 24.04.2 LTS
- Toolchain: zephyr 0.17.0
- Commit SHA: 617b71bc174be034b6b910792ea922c47b568068

ExtremeGTX, 2025-08-25:
@erwango coming from commit zephyrproject-rtos/zephyr@d84348fbd1a17824b6345de814f16bba0c6bc67d

erwango, 2025-08-26:
> [@erwango](https://github.com/erwango) coming from commit [d84348f](https://github.com/zephyrproject-rtos/zephyr/commit/d84348fbd1a17824b6345de814f16bba0c6bc67d)

@marwaiehm-st FYI

marwaiehm-st, 2025-08-28:
Hi @ExtremeGTX,

I have identified that the regression was introduced by commit [cd17871](https://github.com/zephyrproject-rtos/zephyr/commit/cd17871cfe3ae3888569c6989a536b300c410e65), which modified the speed selection logic in the STM32 USB device controller driver.

I would like to highlight that this regression is addressed by the changes in this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/89866). This update refactors the speed selection logic to more accurately consider both hardware support and device configuration, and resolves the USB malfunction observed on STM32F769i_disco.

Once merged, it should resolve the issue.

closed 2025-10-01 by mathieuchopstm

PeeJay, 2025-11-10:
Not sure if it's relevant but I have been having trouble with `<wrn> udc: Spurious resume event` on STM32G4. Enabling `CONFIG_UDC_DRIVER_LOG_LEVEL_ERR=y` remarkably fixes the error.

ExtremeGTX, 2026-01-19:
```
[  725.386346] usb 2-1: new high-speed USB device number 7 using ehci-pci
[  725.634144] usb 2-1: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 4.03
[  725.634151] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  725.634153] usb 2-1: Product: USBD MSC sample
[  725.634155] usb 2-1: Manufacturer: Zephyr Project
[  725.634157] usb 2-1: SerialNumber: 33373330313851020023003B
[  725.639869] usb-storage 2-1:1.0: USB Mass Storage device detected
[  725.640512] scsi host33: usb-storage 2-1:1.0
[  726.678164] scsi 33:0:0:0: Direct-Access     Zephyr   FlashDisk        0.00 PQ: 0 ANSI: 4
[  726.679276] sd 33:0:0:0: Attached scsi generic sg3 type 0
[  726.690213] sd 33:0:0:0: [sdb] 126976 512-byte logical blocks: (65.0 MB/62.0 MiB)
[  726.696073] sd 33:0:0:0: [sdb] Write Protect is off
[  726.696098] sd 33:0:0:0: [sdb] Mode Sense: 03 00 00 00
[  726.703094] sd 33:0:0:0: [sdb] No Caching mode page found
[  726.703100] sd 33:0:0:0: [sdb] Assuming drive cache: write through
[  726.750119]  sdb:
[  726.750229] sd 33:0:0:0: [sdb] Attached SCSI removable disk
[  727.443486] FAT-fs (sdb): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  765.751395] usb 2-1: reset high-speed USB device number 7 using ehci-pci
[  767.054058] kauditd_printk_skb: 146 callbacks suppressed
[  767.054061] audit: type=1400 audit(1768862934.974:158): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=5411 comm="apparmor_parser"
[  767.056835] audit: type=1400 audit(1768862934.977:159): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=5411 comm="apparmor_parser"
[  767.069624] audit: type=1400 audit(1768862934.989:160): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd//third_party" pid=5411 comm="apparmor_parser"
[  768.654654] audit: type=1400 audit(1768862936.584:161): apparmor="DENIED" operation="capable" class="cap" profile="/usr/sbin/cupsd" pid=5713 comm="cupsd" capability=12  capname="net_admin"
[  788.966570] workqueue: blk_mq_run_work_fn hogged CPU for >10000us 11 times, consider switching to WQ_UNBOUND
[  796.417189] usb 2-1: reset high-speed USB device number 7 using ehci-pci
[  813.369692] usb 2-1: reset high-speed USB device number 7 using ehci-pci
[  813.857561] usb 2-1: reset high-speed USB device number 7 using ehci-pci
[  824.622487] usb 2-1: reset high-speed USB device number 7 using ehci-pci
[  824.874867] sd 33:0:0:0: Device offlined - not ready after error recovery
[  824.874896] sd 33:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_ABORT driverbyte=DRIVER_OK cmd_age=89s
[  824.874914] sd 33:0:0:0: [sdb] tag#0 CDB: Write(10) 2a 00 00 00 00 68 00 00 f0 00
[  824.874917] I/O error, dev sdb, sector 104 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.874999] sd 33:0:0:0: rejecting I/O to offline device
[  824.875003] I/O error, dev sdb, sector 344 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875086] I/O error, dev sdb, sector 584 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875124] I/O error, dev sdb, sector 824 op 0x1:(WRITE) flags 0x4800 phys_seg 3 prio class 0
[  824.875135] I/O error, dev sdb, sector 1064 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875143] I/O error, dev sdb, sector 1304 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875151] I/O error, dev sdb, sector 1544 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875159] I/O error, dev sdb, sector 1784 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875167] I/O error, dev sdb, sector 2024 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875174] I/O error, dev sdb, sector 2264 op 0x1:(WRITE) flags 0x4800 phys_seg 2 prio class 0
[  824.875511] Buffer I/O error on dev sdb, logical block 1, lost async page write
[  824.875536] Buffer I/O error on dev sdb, logical block 2, lost async page write
[  824.875538] Buffer I/O error on dev sdb, logical block 3, lost async page write
[  824.875539] Buffer I/O error on dev sdb, logical block 4, lost async page write
[  824.875541] Buffer I/O error on dev sdb, logical block 5, lost async page write
[  824.875542] Buffer I/O error on dev sdb, logical block 6, lost async page write
[  824.875763] Buffer I/O error on dev sdb, logical block 64, lost async page write
[  825.011763] FAT-fs (sdb): Directory bread(block 64) failed
[  825.011820] FAT-fs (sdb): Directory bread(block 64) failed
[  855.768963] blk_print_req_error: 45 callbacks suppressed
[  855.768966] I/O error, dev sdb, sector 64 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  855.768978] FAT-fs (sdb): unable to read inode block for updating (i_pos 1028)
```

I tried again, it mounts successfully on Linux, but on Windows "Explorer" keeps loading forever.
On Linux, I tried to copy a big file and as you see I got a lot of I/O error, but I'm not sure now if it is STM32 UDC Driver or something else, if you think it is irrelevant, feel free to close the Bug.

Thanks for your help.