zephyrproject-rtos/zephyr #94934: stm42f412 "Failed to enable SDMMC domain clock"

fariouche, 2025-08-25:
### Describe the bug

Hello,

Using zephyr 4.2, I have an error when calling disk_access_init (more precisely the ioctl version) and I get the error "Failed to enable SDMMC domain clock".
Disk name is "sdio1".
Of course, the dts node is enabled by default, and I have the config DISK_ACCESS and SDMMC_STM32

A workaround is to remove the second clock in the sdmmc dts node like in my overlay:
```
&sdmmc1 {
	clocks = <&rcc STM32_CLOCK(APB2, 11U)>;
};
```

I don't know the implications of that or what I'm missing to avoir the above error

Thanks

### Regression

- [ ] This is a regression.

### Steps to reproduce

Just call `disk_access_ioctl("sdio1" DISK_IOCTL_CTRL_INIT, NULL);`
Using an stm32f412_discovery board

### Relevant log output

```shell
E: Failed to enable SDMMC domain clock
E: failed to init clocks
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

erwango, 2025-08-26:
Ok, the issue seems legit indeed, we'll have a look.

FRASTM, 2025-09-19:
@fariouche 
I found STM32_CK48_ENABLED is not present in this series ; can you please check with the following : 
```patch
diff --git a/include/zephyr/drivers/clock_control/stm32_clock_control.h b/include/zephyr/drivers/clock_control/stm32_clock_control.h
index 6d342fd2149..e05dbb9cff6 100644
--- a/include/zephyr/drivers/clock_control/stm32_clock_control.h
+++ b/include/zephyr/drivers/clock_control/stm32_clock_control.h
@@ -220,6 +220,10 @@
 #define STM32_PLLI2S_R_DIVISOR		DT_PROP_OR(DT_NODELABEL(plli2s), div_r, 1)
 #endif
 
+#if DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(clk48), st_stm32_clock_mux, okay)
+#define STM32_CK48_ENABLED	1
+#endif
+
 #if DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(pllsai1), st_stm32l4_pllsai_clock, okay)
 #define STM32_PLLSAI1_ENABLED	1
```
(I do not have the same stm32 board )