zephyrproject-rtos/zephyr #94954: Spurious error log "No valid legacy adv to resume" in bt_hci_le_enh_conn_complete when using bt_le_ext_adv_start()

stsai666, 2025-08-25:
### Describe the bug

## Description

When using `bt_le_ext_adv_start()` to start advertising, an error message "No valid legacy adv to resume" is logged by `bt_hci_le_enh_conn_complete()` during normal connection establishment. This error message appears even though the system is functioning correctly, causing confusion for developers.

## Expected Behavior

No error messages should be logged during normal operation when using advertising APIs correctly.

## Actual Behavior

The following error message appears in logs when a connection is established after starting advertising:
```
[00:00:00.123,456] <err> bt_adv: No valid legacy adv to resume
```

## Root Cause Analysis

The issue was introduced in commit [`39cb574438f`](https://github.com/zephyrproject-rtos/zephyr/commit/39cb574438fb29a68d187105053094509d4cd6ed) which changed the log level from `LOG_DBG` to `LOG_ERR` in `bt_le_adv_resume()` function.

### Code Flow:
1. Application calls `bt_le_ext_adv_start()` to start advertising
2. Extended advertising does not create a legacy advertiser instance (`adv_create_legacy()`)
3. When a connection is established, `bt_hci_le_enh_conn_complete()` is called
4. If controller supports `BT_LE_STATES_PER_CONN_ADV`, `bt_le_adv_resume()` is called unconditionally ([line 1500 in `hci_core.c`](https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/subsys/bluetooth/host/hci_core.c#L1500))
5. `bt_le_adv_resume()` calls `bt_le_adv_lookup_legacy()` which returns NULL for extended advertising
6. The error message is logged even though this is expected behavior

### Relevant Code Locations:

**[zephyr/subsys/bluetooth/host/hci_core.c:1494-1501](https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/subsys/bluetooth/host/hci_core.c#L1494-L1501)**
```c
/* if the controller supports, lets advertise for another
 * peripheral connection.
 * check for connectable advertising state is sufficient as
 * this is how this le connection complete for peripheral occurred.
 */
if (BT_LE_STATES_PER_CONN_ADV(bt_dev.le.states)) {
    bt_le_adv_resume();
}
```

**[zephyr/subsys/bluetooth/host/adv.c:1531-1534](https://github.com/zephyrproject-rtos/zephyr/blob/v4.2.0/subsys/bluetooth/host/adv.c#L1531-L1534)**
```c
if (!adv) {
    LOG_ERR("No valid legacy adv to resume");
    return;
}
```

## Steps to Reproduce

1. Configure Zephyr with extended advertising support (`CONFIG_BT_EXT_ADV=y`)
2. Create an advertising set using `bt_le_ext_adv_create()`
3. Start advertising using `bt_le_ext_adv_start()`
4. Establish a connection from a central device
5. Observe the error log during connection establishment

## Environment

- **Zephyr Version**: v4.2.0+ (commit [`39cb574438f`](https://github.com/zephyrproject-rtos/zephyr/commit/39cb574438fb29a68d187105053094509d4cd6ed) and later)
- **Architecture**: Any (issue is in host stack)
- **Board**: Any with Bluetooth LE support and controller that supports `BT_LE_STATES_PER_CONN_ADV`

## Proposed Solutions

### Option 1: Improve Logic in bt_hci_le_enh_conn_complete()
Modify `bt_hci_le_enh_conn_complete()` to only call `bt_le_adv_resume()` when there's actually a legacy advertiser to resume:

```c
if (BT_LE_STATES_PER_CONN_ADV(bt_dev.le.states)) {
    struct bt_le_ext_adv *adv = bt_le_adv_lookup_legacy();
    if (adv && atomic_test_bit(adv->flags, BT_ADV_PERSIST) &&
        !atomic_test_bit(adv->flags, BT_ADV_ENABLED)) {
        bt_le_adv_resume();
    }
}
```

This approach prevents the unnecessary call to `bt_le_adv_resume()` when using extended advertising.

### Option 2: Revert Log Level (Minimal Fix)
Change the log level back to `LOG_DBG` in `bt_le_adv_resume()` since the absence of a legacy advertiser is expected when using extended advertising.

```c
if (!adv) {
    LOG_DBG("No valid legacy adv to resume");
    return;
}
```

### Option 3: Enhanced Error Handling (Alternative)
Add context-aware logging that distinguishes between actual errors and expected behavior:

```c
if (!adv) {
    if (IS_ENABLED(CONFIG_BT_EXT_ADV) &&
        BT_DEV_FEAT_LE_EXT_ADV(bt_dev.le.features)) {
        LOG_DBG("No legacy adv to resume (extended advertising in use)");
    } else {
        LOG_ERR("No valid legacy adv to resume");
    }
    return;
}
```

## Impact

- **Severity**: Low (cosmetic issue, no functional impact)
- **User Experience**: Confusing error messages in logs during normal operation
- **Debugging**: Makes it harder to identify actual issues due to noise in logs

## Additional Context

The original intent of commit [`39cb574438f`](https://github.com/zephyrproject-rtos/zephyr/commit/39cb574438fb29a68d187105053094509d4cd6ed) was to aid in debugging by making advertising resume failures more visible. However, the change didn't account for the fact that extended advertising scenarios legitimately don't have legacy advertisers to resume.

This issue affects any application using the extended advertising APIs (`bt_le_ext_adv_*`) in Zephyr 4.2.0 and later versions.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Configure Zephyr with extended advertising support (`CONFIG_BT_EXT_ADV=y`)
2. Create an advertising set using `bt_le_ext_adv_create()`
3. Start advertising using `bt_le_ext_adv_start()`
4. Establish a connection from a central device
5. Observe the error log during connection establishment

### Relevant log output

```shell
[00:05:00.875,000] <err> bt_adv: No valid legacy adv to resume
[00:05:00.876,000] <inf> app: connected
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

Zephyr Version: v4.2.0+ (commit [39cb574438f](https://github.com/zephyrproject-rtos/zephyr/commit/39cb574438fb29a68d187105053094509d4cd6ed) and later)
Architecture: Any (issue is in host stack)
Board: Any with Bluetooth LE support and controller that supports BT_LE_STATES_PER_CONN_ADV

### Additional Context

The original intent of commit [39cb574438f](https://github.com/zephyrproject-rtos/zephyr/commit/39cb574438fb29a68d187105053094509d4cd6ed) was to aid in debugging by making advertising resume failures more visible. However, the change didn't account for the fact that extended advertising scenarios legitimately don't have legacy advertisers to resume.

This issue affects any application using the extended advertising APIs (bt_le_ext_adv_*) in Zephyr 4.2.0 and later versions.

alwa-nordic, 2025-09-03:
Thank you for analyzing this! We should revert https://github.com/zephyrproject-rtos/zephyr/commit/39cb574438fb29a68d187105053094509d4cd6ed as the quick fix. Would like to open a PR for this?

Automatic advertiser resuming has been deprecated, and the deprecation period is over now. We should start removing all code to do with this, including `bt_le_adv_resume` now. Let's just do that instead of improving it.

stsai666, 2025-09-03:
I've opened PR #95382 to revert 39cb574 as a quick fix. Thanks!

PR #95382 (2025-09-05): Bluetooth: host: adv: revert 39cb574 to fix spurious error log
Commit 39cb574 changed the log level from LOG_DBG to LOG_ERR in bt_le_adv_resume(). This causes the error log
"No valid legacy adv to resume" to appear during normal connection establishment when using bt_le_ext_adv_start(), even though the system is functioning correctly.

Revert the change to restore the original LOG_DBG level.

Fixes: #94954
commit 07dc0ad4 (2025-09-05): Bluetooth: host: adv: revert 39cb574 to fix spurious error log

Commit 39cb574 changed the log level from LOG_DBG to LOG_ERR in
bt_le_adv_resume(). This causes the error log
"No valid legacy adv to resume" to appear during normal connection
establishment when using bt_le_ext_adv_start(), even though the
system is functioning correctly.

Revert the change to restore the original LOG_DBG level.

Fixes: #94954
PR #95885 (2025-09-30): [Backport v4.2-branch] Bluetooth: host: adv: revert 39cb574 to fix spurious error log
Backport 07dc0ad40d37907fb37bffa025226606b6232416 from #95382.

Fixes #94954
commit 116c4a9e (2025-09-30): Bluetooth: host: adv: revert 39cb574 to fix spurious error log

Commit 39cb574 changed the log level from LOG_DBG to LOG_ERR in
bt_le_adv_resume(). This causes the error log
"No valid legacy adv to resume" to appear during normal connection
establishment when using bt_le_ext_adv_start(), even though the
system is functioning correctly.

Revert the change to restore the original LOG_DBG level.

Fixes: #94954

(cherry picked from commit 07dc0ad40d37907fb37bffa025226606b6232416)
closed 2025-09-05 by henrikbrixandersen