zephyrproject-rtos/zephyr #94959: Bluetooth: host: Public access to bt_addr_le_is_resolved()

koffes, 2025-08-25:
### Summary

Currently, this function resides https://github.com/zephyrproject-rtos/zephyr/blob/c4ddc85e6957506b20dff212187fc7a2efeecef8/subsys/bluetooth/host/addr_internal.h#L14

It would be great to have access to this function through a public API.

Ref discussion with @Thalley .

### Describe the solution you'd like

The declaration of `bt_addr_le_is_resolved()` should be in a public header.

jhedberg, 2025-08-26:
No objections. Feel free to submit a PR.

alwa-nordic, 2025-08-26:
I strongly suspect this function does not do what you think it does :)

Please explain what the function you want should do so we are all on the same page and can discuss adding it.

Thalley, 2025-08-26:
> I strongly suspect this function does not do what you think it does :)
> 
> Please explain what the function you want should do so we are all on the same page and can discuss adding it.

Was about to add something similar here:
One issue with this, is that the `addr->type` exposed to the application layer, are usually (always?) either `BT_ADDR_LE_PUBLIC` or `BT_ADDR_LE_RANDOM`, so using this on a public facing `bt_addr_le_t` may always be false.

alwa-nordic, 2025-08-26:
https://github.com/zephyrproject-rtos/zephyr/pull/95011: I'm adding documentation to `bt_addr_le_is_resolved()` to explain what it does.

closed 2025-08-27 by Thalley