zephyrproject-rtos/zephyr #95004: Bluetooth: Round-up BT_GAP_US_TO_CONN_TIMEOUT to 10 ms unit

cvinayak, 2025-08-26:
### Summary

Round-up BT_GAP_US_TO_CONN_TIMEOUT to 10 ms unit.

Given a connection interval in microseconds, round-up the value to 10 ms unit.

Check to ensure BT Spec required `(1 + connPeripheralLatency) × connSubrateFactor × connInterval × 2` should be in the Host Stack APIs, hence this define rounding up the value to 10 ms unit is sufficient.

If this existing define should round up or have a new defined for a round-up value in 10ms units is open for comments.

### Describe the solution you'd like

If this existing define should round up or have a new defined for a round-up value in 10ms units is open for comments.

Thalley, 2025-08-26:
But isn't it odd to e.g. round down `BT_GAP_MS_TO_CONN_INTERVAL(7)` but then round up `BT_GAP_US_TO_CONN_TIMEOUT(7 * 6)`?

Thalley, 2025-08-26:
Sure, since there's a relationsship between the interval and the timeout 

> The Supervision_Timeout[i] in milliseconds shall be larger than (1 + Max_Latency[i]) × Connection_Interval_Max[i] × 2, where Connection_Interval_Max[i] is given in milliseconds (see [Vol 6] Part B, Section 4.5.2).

Then you provide e.g. `BT_GAP_MS_TO_CONN_INTERVAL(11)` (rounded down to 10ms) and Max_Latency=0, then the you would expect `BT_GAP_US_TO_CONN_TIMEOUT(22)` to be a fitting value (since it's twice that of the connection interval. However if you round `BT_GAP_US_TO_CONN_TIMEOUT` up to 30ms and `BT_GAP_MS_TO_CONN_INTERVAL` down to 10ms, then it's 3 times the max connection interval. 

Having `BT_GAP_MS_TO_CONN_INTERVAL` round up to 11.250ms does not "fix" it, but it seems to make more sense, as it closer to the intended. 

If they both round down, then the user gets exactly what's expected for those particular values, but of course with `BT_GAP_MS_TO_CONN_INTERVAL(12)` (rounded down to 11.250ms) and `BT_GAP_US_TO_CONN_TIMEOUT(24)` (rounded down to 20ms) won't work. 

This is of course tricky to do in a generic way, given that each of these macros have very different units in the HCI command, so I don't think that performing any specific type of rounding will be a perfect solution, but I do tend to agree that it makes sense to round up the values that have a dependency like the connection timeout, so I'm not against this change.

closed 2026-09-07 by github-actions[bot]