zephyrproject-rtos/zephyr #95028: frdm_mcxn947:  tests/drivers/watchdog/wdt_basic_reset_none/drivers.watchdog.reset_none_ewm FAILED Failed harness:'Testsuite failed'

pkunieck, 2025-08-26:
### Describe the bug

Tests fails with error massage Failed harness:'Testsuite failed'. 

### Regression

- [ ] This is a regression.

### Steps to reproduce

twister -p frdm_mcxn947/mcxn947/cpu0 -T tests/drivers/watchdog/wdt_basic_reset_none/ -ll DEBUG

### Relevant log output

```shell
INFO    - 306/306 frdm_mcxn947/mcxn947/cpu0 drivers.watchdog.reset_none_ewm                    FAILED Failed harness:'Testsuite failed' (device: 52LSCMSXFJW3W, 11.830s <zephyr>)
INFO    - /__w/zephyrproject/twister-out/frdm_mcxn947_mcxn947_cpu0/zephyr/tests/drivers/watchdog/wdt_basic_reset_none/drivers.watchdog.reset_none_ewm/handler.log
ERROR   - ***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build v4.2.0-2330-g23e0d30a39c7 (delayed boot 500ms) ***
Running TESTSUITE wdt_basic_reset_none
===================================================================
START - test_wdt_bad_window_max
 PASS - test_wdt_bad_window_max in 0.001 seconds
===================================================================
START - test_wdt_callback_reset_none
Feeding watchdog 2 times
Feeding 1
Feeding 2

    Assertion failed at WEST_TOPDIR zephyrproject/tests/drivers/watchdog/wdt_basic_reset_none/src/main.c:98: test_wdt_callback_reset_none: (wdt_interrupted_flag not equal to WDT_TEST_CB_TEST_VALUE)
wdt callback failed
 FAIL - test_wdt_callback_reset_none in 0.065 seconds
===================================================================
TESTSUITE wdt_basic_reset_none failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [wdt_basic_reset_none]: pass = 1, fail = 1, skip = 0, total = 2 duration = 0.066 seconds
 - PASS - [wdt_basic_reset_none.test_wdt_bad_window_max] duration = 0.001 seconds
 - FAIL - [wdt_basic_reset_none.test_wdt_callback_reset_none] duration = 0.065 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: d68efb06ae238cca3c7b7bce63491082
PROJECT EXECUTION FAILED
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

OS: Linux
Commit SHA or Version used: zephyr-v4.2.0-2330-g23e0d30a39c7

majunkier, 2025-09-03:
Test is passing, but i don't have details what was initial problem.

arikgreen, 2025-09-19:
at todays run the test passed
`INFO    - 207/307 frdm_mcxn947/mcxn947/cpu0 drivers.watchdog.reset_none_ewm                    PASSED`

yvesll, 2025-11-03:
Hi @pkunieck can you attach more info like the twister run context. I have tried multiple times and no failure.

closed 2026-01-17 by github-actions[bot]