zephyrproject-rtos/zephyr #95038: Support for Synopsys Advanced I2C/SMBus Controller (H464-0)

taehyun9203, 2025-08-27:
### Summary

Hi,

I have a question regarding the i2c_dw.c driver.
Currently, the register offsets used in the driver seem to be based on an older DesignWare I2C version (around 2015). These offsets do not match with the recent Synopsys I2C/SMBus controller.

In particular, the product code we are working with is H464-0 (Advanced I2C/SMBus controller from Synopsys), which has a different register layout and additional features.

Would it be possible to update or extend the driver to support this newer controller?
If needed, I can help by providing the register map and testing patches on hardware.

Thanks in advance!

### Describe the solution you'd like

N/A

### Alternatives

N/A

### Additional Context

N/A

SinghCod3r, 2025-08-31:
@henrikbrixandersen , @taehyun9203 , can you assign this to me i want to work on this.

SinghCod3r, 2025-09-01:
For Assigning Part? you just have to see right top corner there will be assignees written. you just have to hover there and click on it. You will see list on people and select mine if you want to assign me. You will see my username as SInghCod3r. You click on that and then enter. That's it, you have assigned me the project. Please take a moment and understand this.

taehyun9203, 2025-09-01:
> For Assigning Part? you just have to see right top corner there will be assignees written. you just have to hover there and click on it. You will see list on people and select mine if you want to [...]

I don’t seem to have permission to assign this issue.  
Could a maintainer please assign it to the appropriate person?  
Thanks!

closed 2026-09-06 by github-actions[bot]