zephyrproject-rtos/zephyr #95040: Modem cellular and L2 PPP fsm stuck

davidk88, 2025-08-27:
### Describe the bug

Hello,

I'm using custom board with Quectel EG915 modem and latest zephyr. I'm trying to implement "always connected" algorithm (regain connectivity in case of loss).

So first questiion. What is the idea with zephyr's cellular subsys to regain connectivity in case of L4_DISCONNECTED event (or should i observe any other event)? Is the idea to suspend and resume modem? 

When the network is lost (I'm simulating with cancel location (from SIM card provider portal). I'm trying with modem suspend/resume (pm_device_action_run), but the PPP fsm is stuck in state PPP_STOPPED:

```
[00:07:29.597,000] conn_mgr: conn_mgr_ipv4_events_handler: (0x2000bb50): IPv4 event 0xe030000000000002 received on iface 1 (0x20002ee0)
[00:07:29.597,000] conn_mgr: conn_mgr_ipv4_events_handler: (0x2000bb50): Iface index 0
[00:07:29.597,000] m_conn: event_handler: Network disconnected event
[00:07:29.597,000] net_ctx: net_context_unref: (0x20002fb8): Context 0x2000bbf8 released
[00:07:29.647,000] m_conn: conn_thread: L4 reconnect procedure!
[00:07:29.647,000] conn_mgr: conn_mgr_iface_events_handler: (0x2000bb50): Iface event 0xd010000000000001 received on iface 1 (0x20002ee0)
[00:07:29.647,000] conn_mgr: conn_mgr_iface_events_handler: (0x2000bb50): Iface index 0
Here i suspend and the resume modem:
[00:07:31.683,000] m_conn: conn_thread: modem suspend: 0
[00:07:36.683,000] m_conn: conn_thread: modem resume: 0

[00:07:50.709,000] conn_mgr: conn_mgr_iface_events_handler: (0x2000bb50): Iface event 0xd010000000000002 received on iface 1 (0x20002ee0)
[00:07:50.709,000] conn_mgr: conn_mgr_iface_events_handler: (0x2000bb50): Iface index 0
[00:07:50.709,000] net_l2_ppp: ppp_change_phase_debug: (0x2000bb50): [0x2000c278] phase DEAD (0) => ES
TABLISH (1) (ppp_open_async():221)
[00:07:50.709,000] net_l2_ppp: ppp_open_async: (0x2000bb50): Starting LCP
[00:07:50.709,000] net_l2_ppp: ppp_fsm_lower_up: (0x2000bb50): [LCP/0x2000c2b0] Current state STOPPED (3)
[00:07:50.709,000] net_l2_ppp: ppp_fsm_lower_up: (0x2000bb50): [LCP/0x2000c2b0] Invalid state STOPPED (3)
[00:07:50.709,000] net_l2_ppp: ppp_fsm_open: (0x2000bb50): [LCP/0x2000c2b0] Current state STOPPED (3)
```

At this point fsm is stuck, unless I do suspend/resume again. Then the connectivity gets up. Also in this stuck state if I want to bring iface down via shell, error is returned: Cannot take interface 1 down (-11).

in file `zephyr/subsys/net/l2/fsm.c` in function `ppp_fsm_init` there is a line `fsm->flags = 0U` and in `ppp_fsm_open` where PPP_STOPPED is checked there is a `if (fsm->flags & FSM_RESTART)`. If i set `fsm->flags = FSM_RESTART` in `ppp_fsm_init` then the fsm don't get stuck. But probably there is a reason why flags are set to 0.

Somehow related: https://github.com/zephyrproject-rtos/zephyr/issues/75603

Best regards

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Showstopper – Prevents release or major functionality; system unusable.

davidk88, 2025-11-11:
Please remove the stale label.

MTsimon, 2025-12-04:
I can confirm that this issue can be reproduced on the current main branch.

For additional context:
CONFIG_NET_L2_PPP_TIMEOUT appears to influence how the FSM transitions to the CLOSED or STOPPED states. When the carrier is lost, a Termination-Request must be processed and ACKed/NACKed. I have observed that if higher-level functions, such as dns_resolve(), are not cancelled before the timeout specified by CONFIG_NET_L2_PPP_TIMEOUT, the PPP FSM transitions to the STOPPED state. In this state, it cannot start again unless the FSM_RESTART flag is set.

Like @davidk88, I am also very curious about the exact purpose of the restart flag and how it is intended to be set. Why not always treat a second OPEN as a DOWN/UP sequence?

Once PPP enters the STOPPED state, I cannot see any way for higher-level functions to move it back to the CLOSED state.

Below is a log showing how I managed to get the connection working again by cancelling the ongoing DNS request as soon as L2 notifies that the link is lost. If I do not cancel the request, I end up in the STOPPED state, similar to what @davidk88 described.

```
[00:00:41.690,063] <dbg> net_l2_ppp: net_pkt_hexdump: send L2
[00:00:41.690,155] <dbg> net_l2_ppp: net_pkt_hexdump: 0x200358f8
                                     45 00 00 38 00 00 00 00  40 11 e3 e0 64 46 22 7f |E..8.... @...dF".
                                     08 08 08 08 df ab 00 35  00 24 20 1d b3 b5 01 00 |.......5 .$ .....
                                     00 01 00 00 00 00 00 00  03 64 6e 73 06 67 6f 6f |........ .dns.goo
                                     67 6c 65 00 00 01 00 01                          |gle.....
[00:00:41.852,416] <dbg> net_l2_ppp: net_pkt_hexdump: recv L2
[00:00:41.852,508] <dbg> net_l2_ppp: net_pkt_hexdump: 0x20035cb8
                                     c0 21 05 04 00 04                                |.!....
[00:00:41.852,600] <dbg> net_l2_ppp: ppp_fsm_input: (rx_q[0]): [LCP/0x2000a4b0] LCP Terminate-Req (5) id 4 payload len 0
[00:00:41.852,661] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x2000a4b0] Current state OPENED (9)
[00:00:41.852,691] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x2000a4b0] Terminated by peer
[00:00:41.852,813] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [LCP/0x2000a4b0] state OPENED (9) => STOPPING (5) (fsm_recv_terminate_req():878)
[00:00:41.852,874] <dbg> net_l2_ppp: ppp_fsm_lower_down: (rx_q[0]): [IPCP/0x2000a580] Current state OPENED (9)
[00:00:41.852,996] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [IPCP/0x2000a580] state OPENED (9) => STARTING (1) (ppp_fsm_lower_down():282)
[00:00:41.853,149] <dbg> net_l2_ppp: ppp_change_phase_debug: (rx_q[0]): [0x2000a478] phase RUNNING (4) => TERMINATE (5) (ppp_network_down():37)
[00:00:41.853,210] <dbg> net_l2_ppp: ppp_network_down: (rx_q[0]): [0x2000a478] Proto IPv4 (0x0021) down (0)
[00:00:41.853,271] <dbg> net_l2_ppp: ppp_fsm_close: (rx_q[0]): [IPCP/0x2000a580] Current state STARTING (1)
[00:00:41.853,393] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [IPCP/0x2000a580] state STARTING (1) => INITIAL (0) (ppp_fsm_close():240)
[00:00:41.853,454] <dbg> net_l2_ppp: ppp_fsm_lower_down: (rx_q[0]): [PAP/0x2000a658] Current state OPENED (9)
[00:00:41.853,546] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [PAP/0x2000a658] state OPENED (9) => STARTING (1) (ppp_fsm_lower_down():282)
[00:00:41.853,637] <dbg> net_l2_ppp: ppp_change_phase_debug: (rx_q[0]): [0x2000a478] phase TERMINATE (5) => DEAD (0) (ppp_link_down():140)
[00:00:41.853,729] <dbg> net_l2_ppp: ppp_change_phase_debug: (rx_q[0]): [0x2000a478] phase DEAD (0) => ESTABLISH (1) (lcp_down():183)
[00:00:41.853,851] <dbg> net_l2_ppp: ppp_send_pkt: (rx_q[0]): [LCP/0x2000a4b0] Sending 6 bytes pkt 0x200358f8 (options len 0)
[00:00:41.854,003] <dbg> net_l2_ppp: net_pkt_hexdump: send L2
[00:00:41.854,064] <dbg> net_l2_ppp: net_pkt_hexdump: 0x200358f8
                                     c0 21 06 04 00 04                                |.!....
[00:00:41.854,370] <inf> internet_manager: DNS query was canceled or timeout occured
[00:00:41.854,431] <wrn> internet_manager: NET_EVENT_L4_DISCONNECTED
[00:00:41.854,888] <err> internet_manager: DNS lookup failed: -113
[00:00:42.155,395] <dbg> net_l2_ppp: net_pkt_hexdump: send L2
[00:00:42.155,487] <dbg> net_l2_ppp: net_pkt_hexdump: 0x200358f8
                                     46 00 00 20 00 00 00 00  01 02 44 d6 00 00 00 00 |F.. .... ..D.....
                                     e0 00 00 02 94 04 00 00  17 00 08 fe e0 00 00 01 |........ ........
[00:00:42.155,944] <dbg> net_l2_ppp: ppp_change_phase_debug: (internet_manager): [0x2000a478] phase ESTABLISH (1) => TERMINATE (5) (lcp_close():166)
[00:00:42.156,005] <dbg> net_l2_ppp: ppp_fsm_close: (internet_manager): [LCP/0x2000a4b0] Current state STOPPING (5)
[00:00:42.156,127] <dbg> net_l2_ppp: ppp_change_state_debug: (internet_manager): [LCP/0x2000a4b0] state STOPPING (5) => CLOSING (4) (ppp_fsm_close():248)
[00:00:44.853,881] <dbg> net_l2_ppp: ppp_fsm_timeout: (sysworkq): [LCP/0x2000a4b0] Current state CLOSING (4)
[00:00:44.854,003] <dbg> net_l2_ppp: ppp_change_state_debug: (sysworkq): [LCP/0x2000a4b0] state CLOSING (4) => CLOSED (2) (ppp_fsm_timeout():136)
[00:00:44.854,095] <dbg> net_l2_ppp: ppp_change_phase_debug: (sysworkq): [0x2000a478] phase TERMINATE (5) => DEAD (0) (ppp_link_terminated():125)
[00:00:44.854,125] <dbg> net_l2_ppp: ppp_link_terminated: (sysworkq): [0x2000a478] Link terminated
[00:00:44.854,766] <wrn> internet_manager: NET_EVENT_IF_DOWN