zephyrproject-rtos/zephyr #95077: Logging thread name may cause use-after-free

tpambor, 2025-08-27:
### Describe the bug

Given the following configuration: 
- `CONFIG_LOG_MODE_DEFERRED=y`
- `CONFIG_LOG_THREAD_ID_PREFIX=y`
- `CONFIG_THREAD_NAME=y`

When a thread logs a message and is then stopped and freed, if the log message is processed afterward, calling `k_thread_name_get(tid)` in log_task reads already freed memory.

https://github.com/zephyrproject-rtos/zephyr/blob/2dd8570eecea0bae4a11ab30dc42695ec20e723c/subsys/logging/log_output.c#L312-L319

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Set the following configuration 
- `CONFIG_LOG_MODE_DEFERRED=y`
- `CONFIG_LOG_THREAD_ID_PREFIX=y`
- `CONFIG_THREAD_NAME=y`

2. Create a thread, that logs something.
3. Free the thread after it has finished.
4. Run with AddressSanitizer to see report of use-after-free.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- Zephyr SDK 0.17.1 
- Zephyr 4.2.0

nordic-krch, 2025-09-01:
I'm not sure if there is any way to check whether thread is still valid without checking kernel structure (since that may be invalid - e.g. was allocated on heap or stack). @andyross @wrsPeterMitsis  do you know?

tpambor, 2025-09-01:
I think it would be necessary to store the thread name in the header of the log msg when it is used for logging instead of only using the thread ID, which might already be invalid, when deferred processing starts. https://github.com/zephyrproject-rtos/zephyr/blob/c31b9e971e1fd665b5850b42bae94a9818c3fc70/subsys/logging/log_msg.c#L48-L50

tpambor, 2025-09-01:
> Yes, it's an unfortunate consequence of deferred logging + thread ID (=pointer to struct k_thread) being used in the log meta-data + lifetime of the `struct k_thread` representing the thread.
> 
> I don't see any satisfactory solution for this problem:
> 
> * copying the thread name impacts logger performance (speed/buffer size),
> * detecting that the thread has finished (or retaining the necessary information) requires additional bookkeeping,
> * recommending to statically allocate the `struck k_thread` when using deferred logging might be limiting.

I agree. I did work around the problem by calling `log_flush()` before freeing, so that all messages that are still pending are printed before `struct k_thread` is freed. Maybe it could at least be documented to flush log messages before freeing a thread, when using deferred logging + `CONFIG_LOG_THREAD_ID_PREFIX=y`.

nordic-krch, 2025-09-03:
Another option is to handle that on processing side in `log_output.c`. Thread name can be "safely" fetched by checking if CONFIG_THREAD_MAX_NAME_LEN number of characters (until null) in `kthread_name_get()` are `isprint` and only then printing it. It can even be optimized by checking if `tid` is in data section (statically allocated data) and "safe" name fetching would be done only in not. Currently there is `linker_is_in_rodata` so `linker_is_in_data` would need to be added.

ClaCodes, 2025-09-30:
Maybe the logging system should maintain an LRU cache of ref-counted thread-name copies.
- copying is amortized
- no "use after free", because the thread-name copies are ref-counted