zephyrproject-rtos/zephyr #95096: FlexPWM not working as intended

aperea-cc, 2025-08-28:
### Describe the bug

Hello, I'm trying to use the FlexPWM peripheral of the MCXA156 SoC in a FRDM_MCXA156 devboard but I can't get it to fully work.
Whatever I try, the pulse and period seen in the associated output pins do not correspond with the values I am assigning it.
I have found a way to reproduce the issue using the blinky_pwm example. Using the reproduction steps below I only get a small pulse of less than 1ms in pin P3_6 whenever the PWM is reconfigured, but nothing in between. AFAIK the example should be setting a 50% duty cycle signal and that is not what I am seeing.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. `west init repro`
2. `cd repro && west update`
3. Copy the following overlay file to `samples/basic/blinky_pwm/boards/frdm_mcxa156.overlay`:
```
#include <zephyr/dt-bindings/pwm/pwm.h>

/ {
	pwmleds {
		compatible = "pwm-leds";
		pwm_led_0: pwm_led_0 {
			pwms = <&flexpwm0_pwm0 0 PWM_MSEC(1) PWM_POLARITY_NORMAL>;
			label = "PWM LED 0";
		};
	};

	aliases {
		pwm-led0 = &pwm_led_0;
		pwm-0 = &flexpwm0_pwm0;
	};
};
```
3. `cd zephyr && west build -b frdm_mcxa156 samples/basic/blinky_pwm`
4. `west flash`
5. See incorrect behavior when probing P3_6.

### Relevant log output

```shell
[00:00:00.000,069] <dbg> pwm_mcux: pwm_mcux_init: Set prescaler 128, reload mode 2
*** Booting Zephyr OS build v4.2.0-2444-g0d46a934171a ***
PWM-based blinky
Calibrating for channel 0...
[00:00:00.017,179] <err> pwm_mcux: Too long period (750000), adjust pwm prescaler!
[00:00:00.025,412] <err> pwm_mcux: Too long period (375000), adjust pwm prescaler!
[00:00:00.033,645] <err> pwm_mcux: Too long period (187500), adjust pwm prescaler!
[00:00:00.041,879] <err> pwm_mcux: Too long period (93750), adjust pwm prescaler!
Done calibrating; maximum/minimum periods 62500000/7812500 nsec
Using period 62500000
Using period 31250000
Using period 15625000
Using period 7812500
Using period 15625000
Using period 31250000
...
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- Windows 11
- Zephyr SDK 0.17.2
- Zephyr commit 0d46a934171a11d0fe08ea2a615f79a92ac0ab9b

aperea-cc, 2025-09-01:
Actually, no, it goes deeper than that:
- Section 23.5 of the Reference Manual (Module operation in low power modes) specifies that the FlexPWM peripherals are "Static" on _any_ sleep mode (excluding Deep Power Down which just turns off the peripheral)
- "Static" here means that the module is not active (Page 829 of the reference manual)
- Using the `wfi` instruction automatically places the CPU on Sleep mode.
So... If my understanding is correct here, FlexPWM can only be used when the CPU is active.

Is my explanation incorrect? Is there anything we can do here to make the FlexPWM peripheral work even when the CPU is idle?

DerekSnell, 2025-09-02:
Hi @aperea-cc ,
I think your findings are correct.  As you found in the MCXA156 Reference Manual, the FlexPWM feature is not available in Sleep mode.  And this corelates with your finding the FlexPWM behavior changes when the Idle thread exectues WFI.  I will confirm if the FlexPWM can be used in Sleep mode, but this may require changing the clock source to the FlexPWM from the default main_clk.

In the meantime, you would need to avoid the WFI in the Idle thread.  I see in [cpu_idle.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/arm/core/cortex_m/cpu_idle.c), there is a hook to avoid the WFI using the Kconfig [CONFIG_ARM_ON_ENTER_CPU_IDLE_HOOK](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_ARM_ON_ENTER_CPU_IDLE_HOOK).  But this Kconfig is not directly user assignable, so I think it would take some enhancing the Zephyr repo to avoid the WFI in this case.

Another option is to write your application so the Idle thread never executes.  You could have your lowest priority thread never yield/sleep, and that would prevent the Idle thread from executing.

Best regards

closed 2026-01-28 by ZhaoxiangJin