zephyrproject-rtos/zephyr #95104: samples: dumb_http_server :frdm_mcxn947/mcxn947/cpu0: build failure

hakehuang, 2025-08-28:
### Describe the bug

- frdm_mcxn947/mcxn947/cpu0
- frdm_mcxn947/mcxn947/cpu1

## git bisect

f5bbdaf8e7fdb04d11834afc7aa807c5cb47e9a8 is the first bad commit
```
commit f5bbdaf8e7fdb04d11834afc7aa807c5cb47e9a8
Author: Fin Maaß <f.maass@vogl-electronic.com>
Date:   Thu Aug 21 14:00:48 2025 +0200

    drivers: ethernet: phy: select MDIO
    
    devices on a bus are recomended to
    select its protocol, apply this to the eth phys
    on the mdio bus, so that they just work by
    enableing the device in the dts.
    
    Signed-off-by: Fin Maaß <f.maass@vogl-electronic.com>

 drivers/ethernet/phy/Kconfig               | 18 ++++++++++--------
 drivers/ethernet/phy/Kconfig.dm8806        |  2 +-
 drivers/ethernet/phy/Kconfig.microchip_t1s |  2 +-
 drivers/ethernet/phy/Kconfig.tja1103       |  2 +-
 drivers/ethernet/phy/Kconfig.tja11xx       |  2 +-
 5 files changed, 14 insertions(+), 12 deletions(-)
```

### Regression

- [x] This is a regression.

### Steps to reproduce

```
west build -b frdm_mcxn947/mcxn947/cpu0 samples\net\sockets\dumb_http_server
```

### Relevant log output

```shell
In file included from /__w/zephyr/zephyr/include/zephyr/sys/util_macro.h:34,
                 from /__w/zephyr/zephyr/include/zephyr/irq_multilevel.h:16,
                 from /__w/zephyr/zephyr/include/zephyr/devicetree.h:20,
                 from /__w/zephyr/zephyr/include/zephyr/device.h:12,
                 from /__w/zephyr/zephyr/drivers/ethernet/phy/phy_mii.c:11:
/__w/zephyr/zephyr/include/zephyr/device.h:96:41: error: '__device_dts_ord_120' undeclared here (not in a function); did you mean '__device_dts_ord_20'?
   96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_macro.h:204:9: note: in expansion of macro 'Z_COND_CODE_1'
  204 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/devicetree.h:5083:9: note: in expansion of macro 'COND_CODE_1'
 5083 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT),   \
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:69:53: note: in expansion of macro '__DEBRACKET'
   69 | #define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
      |                                                     ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_macro.h:204:9: note: in expansion of macro 'Z_COND_CODE_1'
  204 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_macro.h:427:24: note: in expansion of macro 'COND_CODE_1'
  427 | #define UTIL_AND(a, b) COND_CODE_1(UTIL_BOOL(a), (b), (0))
      |                        ^~~~~~~~~~~
/__w/zephyr/zephyr/drivers/ethernet/phy/phy_mii.c:562:17: note: in expansion of macro 'UTIL_AND'
  562 |         .mdio = UTIL_AND(UTIL_NOT(IS_FIXED_LINK(n)),                     \
      |                 ^~~~~~~~
/__w/zephyr/zephyr/include/zephyr/toolchain/common.h:169:23: note: in expansion of macro '_DO_CONCAT'
  169 | #define _CONCAT(x, y) _DO_CONCAT(x, y)
      |                       ^~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:96:33: note: in expansion of macro '_CONCAT'
   96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:297:37: note: in expansion of macro 'DEVICE_NAME_GET'
  297 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:314:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  314 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/ethernet/phy/phy_mii.c:563:26: note: in expansion of macro 'DEVICE_DT_GET'
  563 |                          DEVICE_DT_GET(DT_INST_BUS(n)))                  \
      |                          ^~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/ethernet/phy/phy_mii.c:578:9: note: in expansion of macro 'PHY_MII_CONFIG'
  578 |         PHY_MII_CONFIG(n);                                              \
      |         ^~~~~~~~~~~~~~
/__w/zephyr/zephyr/twister-out/frdm_mcxn947_mcxn947_cpu1/zephyr/samples/net/sockets/dumb_http_server/sample.net.sockets.dumb_http_server/zephyr/include/generated/zephyr/devicetree_generated.h:28403:47: note: in expansion of macro 'PHY_MII_DEVICE'
28403 | #define DT_FOREACH_OKAY_INST_ethernet_phy(fn) fn(0)
      |                                               ^~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_ethernet_phy'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/__w/zephyr/zephyr/drivers/ethernet/phy/phy_mii.c:588:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
  588 | DT_INST_FOREACH_STATUS_OKAY(PHY_MII_DEVICE)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: ubuntu
- toolchain: v0.17.1
- version: a4fde2d2055dd58c5ea4593ef3a85a84dedd777b

maass-hamburg, 2025-08-28:
cpu0 was fine, only cpu1 had this problem, the eth phy was activated in the dts, but the parent bus mdio dev not.

fix: https://github.com/zephyrproject-rtos/zephyr/pull/95122

hakehuang, 2025-09-25:
@maass-hamburg mcx_n9xx_evk/mcxn947/cpu1 need fixes as well, sorry for missing report in this platform.

PR #95122 (2025-09-02): boards: nxp: frdm_mcxn947: deactivate eth phy on cpu1
deactivate eth phy on cpu1, because
mdio is also only activated on cpu0.

fixes: #95104
closed 2025-09-29 by decsny