zephyrproject-rtos/zephyr #95113: gpio_pin_get_config for STM32 returns wrong flags value for input pins

r2r0, 2025-08-28:
### Describe the bug

For STM32WB55 we obtain wrong `flags` value from `gpio_pin_get_config` when it is called for input pins.
The returned value has both flags set: `GPIO_OUTPUT` and `GPIO_INPUT`.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Call `gpio_pin_get_config` for any pin configured (in DTS) as input.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: Linux
-  Tollchain: Zephyr 0.17.0
-  Zepyr 3.7

### Additional Context

Probably wrong code is:
```
	if (pin_cfg.out_state != 0) {
		flags |= GPIO_OUTPUT_HIGH;
	} else {
		flags |= GPIO_OUTPUT_LOW;
	}
```
and there should be instead:
```
	if (pin_cfg.out_state != 0) {
		flags |= GPIO_OUTPUT_INIT_HIGH;
	} else {
		flags |= GPIO_OUTPUT_INIT_LOW;
	}
```

pebeoo, 2025-09-15:
> I have carefully followed your test scenario on Zephyr LTS 3.7.1 with STM32WB55:
> 
> * I configured the pin as input (with pull-up) in both the device tree and application code.
> * I called gpio_pin_get_config immediately after configuration.
> * Here are the results from my board:
> 
> > * gpio_pin_get_config: flags = 0x00070010
> > * GPIO_INPUT:  0x00010000
> > * GPIO_OUTPUT: 0x00020000
> 
> As you can see, the returned flags include GPIO_INPUT and other relevant flags, but do not include GPIO_OUTPUT. PS: In the gpio.h file, the flags are defined as follows: /** Enables pin as input. */ #define GPIO_INPUT (1U << 16)
> 
> /** Enables pin as output, no change to the output state. */ #define GPIO_OUTPUT (1U << 17)

Hello @kbidani as I understand in response to gpio_pin_get_config you received: 0x00070010, below binary comparison:
flags                  0111 0000 0000 0001 0000
GPIO_INPUT:     0001 0000 0000 0000 0000
 GPIO_OUTPUT: 0010 0000 0000 0000 0000. 
 
So your response includes GPIO_OUTPUT | GPIO_INPUT | GPIO_OUTPUT_INIT_LOW and it is caused by code mentioned by @r2r0, because GPIO_OUTPUT_LOW is GPIO_OUTPUT | GPIO_OUTPUT_INIT_LOW.

closed 2025-09-17 by kartben

r2r0, 2025-09-30:
Is it planned backport to 3.7?