zephyrproject-rtos/zephyr #95128: Add supported boards feature comparison grid/table view

miedlp, 2025-08-28:
### Summary

Use case: Zephyr user is trying to find a second source, or board options, based on a set of SoC/board features.

Currently it is a bit tedious to compare boards based on a given set of features, one has to open each board site to see the table of supported features and then has to compare based on these.

### Describe the solution you'd like

Add a grid/table overview that shows boards and their respective features, for example like this (or transposed):

| Board \ Feature  | Feature 0 | Feature 1 | Feature 2 | Feature 3 | Feature 4 | Feature 5 |
| ------------- | ------------- | ------------- | ------------- | ------------- | ------------- | ------------- |
| Board 0  | no | on-chip | on-board | on-chip | on-chip | on-chip |
| Board 1  | on-board | on-chip | on-board | on-chip | on-chip | on-chip |
| Board 2  | no| no | on-board | on-chip | on-chip | on-chip |
| Board 3  | on-board | no | on-board | on-chip | on-chip | on-chip |
| Board 4  | no | on-chip | on-board | on-chip | no | on-chip |

Input from discussion with @kartben at ZDS'25:

* Could be done as another view on the "Supported Boards" site next to "card" and "compact" view. E.g. "grid" or "table"
*  Filtered Features, which are a must, could be either listed in the leftmost columns (or top rows if transposed), or they could be removed entirely to make the whole grid less crowded.
* All necessary data to build the table should already be generated, so this feature would simply be another representation of the existing information.
* Nice-to-have: Easy "download" of a filtered table to be able to include it in corporate docs/excel sheets for managers/product owners/etc.

### Alternatives

Idea from @kartben: add an "you might also be interested in..."-section with a list of similar boards on the pages of each board, like webshops do for products.

ritesh006, 2025-08-31:
is this issue still relevant?

kartben, 2025-09-01:
> Are you already started working on this topic if not I would like to take

I have made a few tests locally but you are more than welcome to take over. :)

closed 2026-09-06 by github-actions[bot]