zephyrproject-rtos/zephyr #95238: frdm_k64f: phy_mc_ksz8081 fails to configure the PHY properly

VitekST, 2025-08-31:
### Describe the bug

Just out of curiosity, I wanted to try `http_server` on the `frdm_k64f`. Sadly, upon it building and flashing, the `phy_mc_ksz8081` driver seems to struggle to configure the Ethernet PHY properly so that communication over Ethernet is possible. I also tried the `dhcpv4_client` sample with the same result.

The issue was bisected to: (@decsny)
```
commit 02b99470a3903c8ea34c65fb1410092374fa821b
Author: Declan Snyder <declan.snyder@nxp.com>
Date:   Wed Aug 27 12:03:32 2025 -0500

    drivers: ethernet: ksz8081: Save link state for get
    
    Instead of using mdio bus for getting link state, only get it in the
    monitor and save it off for get rate api implementation to use
    
    Signed-off-by: Declan Snyder <declan.snyder@nxp.com>
```

I've also set up frame capture and no frames were observed. Checking out at a parent commit (`043ec3d9eeaf0b4d3d2b0963095d8e9668549f81`) - both examples work as intended.

### Regression

- [x] This is a regression.

### Steps to reproduce

1. Checkout at `02b99470a3903c8ea34c65fb1410092374fa821b`.
2. `west update`
3. `cd samples/net/dhcpv4_client`
4. `west build . -p -b frdm_k64f`
5. power up the target
6. `west flash -r jlink` or with any other relevant runner
7. observe console and behaviour

### Relevant log output

```shell
[00:00:01.103,000] <inf> eth_nxp_enet_mac: Link is down
*** Booting Zephyr OS build v4.2.0-2524-g02b99470a390 ***
[00:00:01.103,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:01.103,000] <inf> net_dhcpv4_client_sample: Start on ethernet: index=1
[00:00:01.603,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:04.106,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:04.106,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:04.106,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:00:04.606,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:07.609,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:10.612,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:13.715,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence

# ...and so on - the line above repeats indefinitely.
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: Arch Linux, `6.12.44-1-lts` (yes, btw i use arch)
- toolchain: Zephyr SDK (`arm-zephyr-eabi`)
- revision of interest: `02b99470a3903c8ea34c65fb1410092374fa821b`

herculanodavi, 2025-09-19:
This regression also affects `mimxrt1060_evk@B/mimxrt1062/qspi`. Same steps to reproduce except the board name. 

```
[00:00:00.603,000] <inf> eth_nxp_enet_mac: Link is down
*** Booting Zephyr OS build v4.2.0-2524-g02b99470a390 ***
[00:00:00.603,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.603,000] <inf> net_dhcpv4_client_sample: Start on ethernet: index=1
[00:00:01.103,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:03.906,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:03.906,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:03.906,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:00:04.406,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:07.809,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:11.312,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
[00:00:14.716,000] <inf> phy_mc_ksz8081: PHY (0) is entering autonegotiation sequence
```

The issue persists on `27fa72163203fa3e2d883cdbfcc1fb7ba469e6ed` (current main), but without the logs:

```
[00:00:00.604,000] <inf> eth_nxp_enet_mac: Link is down
*** Booting Zephyr OS build v4.2.0-3956-g27fa72163203 ***
[00:00:00.604,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.604,000] <inf> net_dhcpv4_client_sample: Start on ethernet: index=1
```

decsny, 2025-10-15:
Appears to have been fixed by e2113ce2c5d2f42ef3505164ad302045b7e45184

decsny, 2025-10-16:
I did not have the ethernet cable plugged in, it's one of those days

closed 2025-10-16 by decsny