zephyrproject-rtos/zephyr #95282: stm32f4_disco bxCAN driver does not set filters properly

gfabiano, 2025-09-01:
### Describe the bug

I tried to compile the driver can counter example for **stm32f4_disco** and I noticed that filters does not works properly. In particular I noticed that the led was not blinking: with STD frames the filters does not works, i tried also to set all 0 to pass all frames:

```
	const struct can_filter change_led_filter = {
		.flags = 0U,
		.id = 0u,
		.mask = 0u
	};
```

At the end I noticed that if I set the flags to `CAN_FILTER_IDE` pass std frames and extend frames. but also on this condition some frames are not filtered setting the mask and id fields.

I tried the same example for **nucleo_h753zi** and it works as expected.
So I think there is some bugs on bxCAN driver implementation but I did not found any issue releted on bxCAN/stm32f4 family.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. west build -b stm32f4_disco zephyr\samples\drivers\can\counter
2. west flash
3. **led not blinks**

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Additional Context

I do not know if on other f4 boards with other micro works

gfabiano, 2025-09-03:
I found out that it is a regression caused by #86676 , @henrikbrixandersen

kbidani, 2025-09-05:
On STM32 microcontrollers with the BxCAN peripheral, CAN1 and CAN2 share the same hardware filter banks.
CAN2 cannot operate independently:
The hardware design requires that CAN1 is enabled and clocked, even if your application only uses CAN2.
The property master-can-reg = <0x40006400>; in the Device Tree for CAN2 tells Zephyr (and the hardware) that CAN2 uses CAN1’s filter hardware (the base address 0x40006400 is for CAN1).
So, always enable CAN1 in the Device Tree (set status = "okay";), even if you only use CAN2 in your application.

gfabiano, 2025-09-05:
Hi, after some reading on the datasheet I know that. But it is not my application, it is the counter example. I think that the boards dts should be fixed to enable also can1 to fulfill this hardware requirement. In particular on **stm32f4_disco** board dts there is only the can2 enabled.