zephyrproject-rtos/zephyr #95370: intel_adsp: cavs: kernel.multiprocessing.smp fails with sdk 0.17.4

majunkier, 2025-09-03:
### Describe the bug

Updating SDK to 0.17.4 introduce failure on intel_adsp/cavs25 in test kernel.multiprocessing.smp. (SDK 0.17.2 was fine)

### Regression

- [ ] This is a regression.

### Steps to reproduce

twister --device-testing --hardware-map  <HW_MAP> -s kernel.multiprocessing.smp -ll debug

### Relevant log output

```shell
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_ipi
DEBUG   - DEVICE: cpu num=4
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/smp/src/main.c:753: smp_test_smp_ipi: (sched_ipi_has_called != 0 is false)
DEBUG   - DEVICE: did not receive IPI.(1)
DEBUG   - DEVICE: FAIL - test_smp_ipi in 0.201 seconds
DEBUG   - DEVICE: ===================================================================
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

-OS: Ubuntu 22.04
-Toolchain: Zephyr
-SDK: 0.17.4

majunkier, 2025-09-05:
works on version: v4.2.0-2909-g8f85d8e62ac5
Closing for now, if issue will back i will reopen bug.

arikgreen, 2025-10-07:
No reproduction. Passed on:
Zephyr version: v4.2.0-5092-g8f60b7446ca3

closed 2025-10-07 by arikgreen