zephyrproject-rtos/zephyr #95398: RFC/process: block merging of boards that have no CI testing what so ever

nordicjm, 2025-09-03:
### Problem Description

It's been noticed that mps4 non-secure board targets have been added with no testing, they do not work, they have never worked, their additions are incomplete and this is completely obscured because of the following in the board's twister files:
```
testing:
  only_tags:
    - trusted-firmware-m
```
this limits tests to only tests that have that tag, and to no surprised the 7 samples in tree and 2 tests completely ignore these boards because of allow lists of because of filters like this:
```
common:
  filter: (CONFIG_TFM_BOARD != "") and CONFIG_ARM_NONSECURE_FIRMWARE
```
which will exclude the broken boards because of the following:
```
warning: BUILD_WITH_TFM (defined at soc/st/stm32/Kconfig.defconfig:84,
soc/nxp/lpc/Kconfig.defconfig:6, modules/trusted-firmware-m/Kconfig.tfm:35, modules/trusted-
firmware-m/Kconfig.tfm:35, soc/nordic/nrf53/Kconfig:352, soc/nordic/Kconfig:67,
soc/adi/max32/Kconfig:74) was assigned the value 'y' but got the value 'n'. Check these unsatisfied
dependencies: (SOC_FAMILY_STM32 || SOC_FAMILY_LPC || (TRUSTED_EXECUTION_NONSECURE && TFM_BOARD != ""
&& ARM_TRUSTZONE_M) || (TRUSTED_EXECUTION_NONSECURE && TFM_BOARD != "" && ARM_TRUSTZONE_M && 0) ||
(SOC_SERIES_NRF53X && SOC_FAMILY_NORDIC_NRF) || SOC_FAMILY_NORDIC_NRF || SOC_FAMILY_MAX32) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BUILD_WITH_TFM and/or look up
BUILD_WITH_TFM in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
```
i.e. `CONFIG_TFM_BOARD` is unset

### Proposed Change (Summary)

Therefore I propose some CI check that fails a status check if no CI builds have been built for newly added boards or board targets to prevent merging of boards that are clearly broken or untested

### Proposed Change (Detailed)

A CI check (somehow) that will fail if a board.yml file is changed and for each of the board targets from that file, at least one has no CI build jobs that have ran

PerMac, 2025-09-03:
Such check makes sense to me. For resolve this particular case I think twister and test_pan.py scripts can be modified to ignore/clear `testing: only_tags` when boards are modified https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/ci/test_plan.py#L232

PerMac, 2025-09-03:
@nashif What is a purpose of "only_tags" entry? I thought it is for CI, to limit testing on a given platform to "components" where the platform differs, to limit redundancy in zephyr's CI. If so, since CI performance is in a root of this issue, I proposed to update another key component in CI, the test_plan.py script. 
My understanding was that a rule is to run all applicable tests on a given board when something is changed in such board files. I just propose to "patch" a current behavior to follow the rule: do not descope majority of tests (those missing required tags) in case a board is modified.
> Those exist for a reason, if a target needs them, they should stay

I am not proposing removing "only_tags" entries. I just meant that maybe we shouldn't take this entry into an account when a board is modified and CI scope is being decided. 

>This needs to work and be restable regardless of what we do in CI, i.e. when running twister locally to verify changes, which does not involve test_plan.py

I agree. So to address this and expand my proposal: maybe we should "hide" usage of "only_tags" behind some extra flag for twister call (e.g. --tag-limited-scope)? So when a user is calling twister for a platform this entry is not used unless the user adds it explicite? And we can use this flag in CI's calls to have the redundancy limitation. It will decouple the CI tuning behavior from twister default usage.

Unless I completely missed the point of "only_tags"...

nashif, 2025-09-03:
only_tags exist because we have cases where we need to target specific tests and do not care about anyhthing else. if you remove this, basically you are openning a barrel of worms and expecting targets to build/pass on things they do not target. This RFC is not about only_tags, that was just the trigger, there is no issue with only_tags.

PerMac, 2025-09-04:
> expecting targets to build/pass on things they do not target

It means that it is by design that platforms with only_tags don't have to pass any other test than those with a given tag? Ok, got it now.

nordicjm, 2025-09-04:
There is this on https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#limited-scope despite it being on the blobs page, it doesn't specifically state with blobs:
```
The project maintainers define an open source test suite that an upstream target must be able to pass using only open source software included in the mainline distribution and its modules. The open source test suite currently consists of:

    samples/philosophers

    tests/kernel

The scope of this test suite may grow over time. The goal is to specify tests for a minimal feature set which must be supported via open source software for any target with upstream Zephyr support.

At the discretion of the release team, the project may remove support for a hardware target if it cannot pass this test suite.
```

So if this could be moved/copied out to the zephyr contribution guide for boards, and a check added to ensure these build and pass, that would be a good fix/improvement

wearyzen, 2025-09-04:
A clear documentation in board porting guide stating that _it is mandatory for boards to build hello world_ would help but before taking a decision please consider the context in which the MPS4 NS board was added.

- Support for MPS4 boards was added in TF-M v2.2.x but Zephyr was not on it and was in process of moving to this version of TF-M.
- To avoid any delay in upstreaming the board, a step by step approach and all the necessary board files were pushed for review.
- There was also a note added in the boards documentation that it need upstream TF-M to work.
- As long as users download the supported TF-M version they can use the board with a single change to enable TF-M compilation with this board.
- And, TF-M compilation with this board was kept disabled only because Zephyr upstream wasn't on the required version.

After Zephyr updated to the TF-M v2.2.x, I have been working on enabling this board upstream but have faced 2 issues:
1. upstream docker was missing few python dependencies that TF-M 2.2.x needed. This is now fixed and the next upstream docker image should have the necessary dependencies.
2. All TF-M's warning are treated as errors and there are a bunch of warnings in TF-M's MPS4 build. I have been working on fixing them as well. But fixing one of these warnings is non-trivial so there is a slight delay.

I would also like to add that MPS4 is regularly tested internally with Zephyr's testsuite, it is just not built/tested in Zephyr because of the above reasons. It is being actively worked on and in a hopeful few days would be enabled in Zephyr ci as well.

So, along with adding any new checks in ci, I would request you all to help and guide on how such boards, that need to follow a step by step approach, be added upstream.

nordicjm, 2025-09-04:
> A clear documentation in board porting guide stating that _it is mandatory for boards to build hello world_ would help but before taking a decision please consider the context in which the MPS4 NS [...]

A board does not need to be added upstream, you can keep your own repo with boards in it and include it in a manifest, then when it is in a ready state whereby things works, can submit it upstream.