zephyrproject-rtos/zephyr #95514: USB sample not working on STM32

everedero, 2025-09-05:
### Describe the bug

Compiling the sample/subsys/usb/testusb sample for an STM32L412 platform on a custom board.
It fails with:
```
*** Booting Zephyr OS build v4.2.0-2937-gac8d17955223 ***
[00:00:00.000,000] <inf> usbd_init: interface 0 alternate 0
[00:00:00.000,000] <inf> usbd_init:     ep 0x81 mps 0x0040 interface ep-bm 0x00020000
[00:00:00.000,000] <inf> usbd_init:     ep 0x01 mps 0x0040 interface ep-bm 0x00020002
[00:00:00.000,000] <inf> usbd_init: interface 1 alternate 0
[00:00:00.000,000] <inf> usbd_init:     ep 0x82 mps 0x0040 interface ep-bm 0x00040000
[00:00:00.000,000] <inf> usbd_init:     ep 0x02 mps 0x0040 interface ep-bm 0x00040004
[00:00:00.000,000] <inf> usbd_init: interface 2 alternate 0
[00:00:00.000,000] <inf> usbd_init:     ep 0x83 mps 0x0000 interface ep-bm 0x00080000
[00:00:00.000,000] <inf> usbd_init:     ep 0x03 mps 0x0000 interface ep-bm 0x00080008
[00:00:00.000,000] <inf> usbd_init: interface 2 alternate 1
[00:00:00.000,000] <inf> usbd_init:     ep 0x83 mps 0x0100 interface ep-bm 0x00080000
[00:00:00.000,000] <err> usbd_init: Failed to assign endpoint addresses
[00:00:00.000,000] <err> usbd_init: Failed to init FS configuration 1
[00:00:00.000,000] <err> main: Failed to initialize device support
```

- It works with Zephyr 3.5, but fails on current 4.2.99 main
- Some samples are still working (like sample/subsys/usb/midi)

### Regression

- [x] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

jfischer-no, 2025-09-16:
> This problem seems related to the test trying to create endpoints with a `wMaxPacketSize` (= 512) larger than maximum allowed:

This board likely does not have enough endpoints. I will fix it in the loopback code.

mathieuchopstm, 2025-09-16:
> > This problem seems related to the test trying to create endpoints with a `wMaxPacketSize` (= 512) larger than maximum allowed:
> 
> [@mathieuchopstm](https://github.com/mathieuchopstm) This board likely does not have enough endpoints. I will fix it in the loopback code.

I don't think so. Running on `b_u585i_iot02a` (which uses the OTG_FS IP; not the ST USB of L412) yields the same error:

<details>
<summary>Error log</summary>

```
*** Booting Zephyr OS build v4.2.0-3409-gb78b0c4f333a ***
[00:00:00.015,000] <inf> usbd_init: interface 0 alternate 0
[00:00:00.021,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.028,000] <inf> usbd_init:     ep 0x81 mps 0x0040 interface ep-bm 0x00020000
[00:00:00.036,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 64
[00:00:00.043,000] <inf> usbd_init:     ep 0x01 mps 0x0040 interface ep-bm 0x00020002
[00:00:00.051,000] <inf> usbd_init: interface 1 alternate 0
[00:00:00.057,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.065,000] <inf> usbd_init:     ep 0x82 mps 0x0040 interface ep-bm 0x00040000
[00:00:00.073,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 64
[00:00:00.080,000] <inf> usbd_init:     ep 0x02 mps 0x0040 interface ep-bm 0x00040004
[00:00:00.088,000] <inf> usbd_init: interface 2 alternate 0
[00:00:00.094,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.101,000] <inf> usbd_init:     ep 0x83 mps 0x0000 interface ep-bm 0x00080000
[00:00:00.109,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 64
[00:00:00.117,000] <inf> usbd_init:     ep 0x03 mps 0x0000 interface ep-bm 0x00080008
[00:00:00.125,000] <inf> usbd_init: interface 2 alternate 1
[00:00:00.131,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.138,000] <inf> usbd_init:     ep 0x83 mps 0x0100 interface ep-bm 0x00080000
[00:00:00.146,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 64
[00:00:00.153,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 64
[00:00:00.161,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 64
[00:00:00.168,000] <err> usbd_init: Failed to assign endpoint addresses
[00:00:00.175,000] <err> usbd_init: Failed to init FS configuration 1
[00:00:00.182,000] <err> main: Failed to initialize device support
```

</details>

After bumping max EP size to 512, the sample starts properly. Notice how it only creates 4 endpoits:

<details>
<summary>Successful start log</summary>

```
*** Booting Zephyr OS build v4.2.0-3409-gb78b0c4f333a ***
[00:00:00.015,000] <inf> usbd_init: interface 0 alternate 0
[00:00:00.021,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.028,000] <inf> usbd_init:     ep 0x81 mps 0x0040 interface ep-bm 0x00020000
[00:00:00.036,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 512
[00:00:00.043,000] <inf> usbd_init:     ep 0x01 mps 0x0040 interface ep-bm 0x00020002
[00:00:00.051,000] <inf> usbd_init: interface 1 alternate 0
[00:00:00.057,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.065,000] <inf> usbd_init:     ep 0x82 mps 0x0040 interface ep-bm 0x00040000
[00:00:00.073,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 512
[00:00:00.080,000] <inf> usbd_init:     ep 0x02 mps 0x0040 interface ep-bm 0x00040004
[00:00:00.088,000] <inf> usbd_init: interface 2 alternate 0
[00:00:00.094,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.102,000] <inf> usbd_init:     ep 0x83 mps 0x0000 interface ep-bm 0x00080000
[00:00:00.110,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 512
[00:00:00.117,000] <inf> usbd_init:     ep 0x03 mps 0x0000 interface ep-bm 0x00080008
[00:00:00.125,000] <inf> usbd_init: interface 2 alternate 1
[00:00:00.131,000] <dbg> udc: ep_check_config: cfg d:I|- t:S|B|I|-, mps 1023
[00:00:00.138,000] <inf> usbd_init:     ep 0x83 mps 0x0100 interface ep-bm 0x00080000
[00:00:00.146,000] <dbg> udc: ep_check_config: cfg d:-|O t:S|B|I|-, mps 512
[00:00:00.154,000] <inf> usbd_init:     ep 0x03 mps 0x0100 interface ep-bm 0x00080008
[00:00:00.162,000] <inf> usbd_init: Instance iface-bm 0x00000007 ep-bm 0x000e000e
[00:00:00.170,000] <inf> usbd_init: Init class node 0x200000dc, descriptor length 100
[00:00:00.178,000] <inf> usbd_init: bNumInterfaces 3 wTotalLength 109
[00:00:00.185,000] <inf> usbd_init: FS bNumConfigurations 1
[00:00:00.191,000] <dbg> udc_stm32: udc_stm32_enable: Enable UDC
[00:00:00.197,000] <dbg> udc_stm32: udc_stm32_mem_init: DRAM size: 1280b
[00:00:00.204,000] <dbg> udc: ep_check_config: cfg d:-|O t:-|-|-|C, mps 64
[00:00:00.212,000] <dbg> udc_stm32: udc_stm32_ep_enable: Enable ep 0x00
[00:00:00.219,000] <dbg> udc: ep_check_config: cfg d:I|- t:-|-|-|C, mps 64
[00:00:00.226,000] <dbg> udc_stm32: udc_stm32_ep_enable: Enable ep 0x80
```

</details>

The L4 Reference Manual says the IP supports up to 8 bi-directional endpoints, so this shouldn't be the issue:
[image]

mathieuchopstm, 2025-10-06:
I don't have hardware to reproduce locally but this should be fixed by #96926.