zephyrproject-rtos/zephyr #95609: doc: Improve documentation of west sdk

ajf58, 2025-09-06:
### Describe the bug

Raising this issue here although there's some overlap with https://github.com/zephyrproject-rtos/west

```
> west --version
West version: v1.4.0
```

```
> west sdk --help
usage: west sdk [-h] <subcommand> ...

List and Install Zephyr SDK

positional arguments:
  <subcommand>  select a subcommand. If omitted, treat it as the 'list' selected.
    list        list installed Zephyr SDKs
...
```
This description is incomplete: `west sdk list` will _also_ lists the available toolchain.

```
> west sdk install --help
...
-t, --toolchains toolchain_name [toolchain_name ...]
                        toolchain(s) to install (e.g. 'arm-zephyr-eabi'). If this option is not given, toolchains for all        
                        architectures will be installed. If you are unsure which one to install, install all toolchains. This    
                        requires downloading several gigabytes and the corresponding disk space.Each Zephyr SDK release may      
                        include different toolchains; see the release notes at https://github.com/zephyrproject-rtos/sdk-        
                        ng/releases.
```
This has a few issues:
1. A small typo (missing whitespace) "...disk space.Each..."
2. Recommending installing every toolchain feels like bad advice. The overwhelming majority of users are working on Arm, RISC-V, or  x86_64; sparc is pretty niche in 2025!
3. It feels like this command should recommend `west sdk list`.
4. `develop/west/built-in.html` doesn't mention `west sdk` at all.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Page: develop/west/built-in
- Version: main
- SHA-1: a4392a31cdbd820a29f06a93cc36aa3a97ae3645

### Additional Context

This issue was reported from the online documentation page using the 'Report an issue' button.

pdgendt, 2026-01-08:
> 4\. `develop/west/built-in.html` doesn't mention `west sdk` at all.

It's not built-in, it's an extension.

marc-hb, 2026-01-08:
Fix submitted in #101966