zephyrproject-rtos/zephyr #95656: JLink: west flash fails to flash STM32G0 when SRAM parity is used

efra-mx-fire, 2025-09-08:
## Pre-requisite:

The running application shall have the SRAM parity enabled:

Flash using the changes specified in https://github.com/zephyrproject-rtos/zephyr/issues/88104

## Issue

West flash fails to flash the target, the  only way to be able to flash it is to first flash it with the STM32 cube.

```
****** Error: Failed to preserve target RAM @ 0x20000000-0x2001FFFF.
Failed to prepare for programming.
Unspecified error -1
```

The command
```
west flash --skip-rebuild -d build/app_cmb -f build/app_cmb/zephyr/merged.hex
```

## Logs

```
west -v flash --skip-rebuild -d build/app_cmb -f build/app_cmb/zephyr/merged.hex
-- west flash: using runner jlink
-- runners.jlink: reset after flashing requested
runners.jlink: JLINKARM_GetDLLVersion()=82200
-- runners.jlink: JLink version: 8.22
runners.jlink: JLink commander script:
ExitOnError 1
r
LE
loadfile "build/app_cmb/zephyr/merged.hex"
r
g
writeDP 1 0
readDP 1
q
-- runners.jlink: Flashing file: build/app_cmb/zephyr/merged.hex
runners.jlink: /opt/SEGGER/JLink_V822/JLinkExe -nogui 1 -if swd -speed 4000 -device STM32G0B1RE -CommanderScript /tmp/tmpcwy6nphrjlink/runner.jlink -nogui 1
SEGGER J-Link Commander V8.22 (Compiled Mar 19 2025 15:52:30)
DLL version V8.22, compiled Mar 19 2025 15:51:26

J-Link Command File read successfully.
Processing script file...
J-Link>ExitOnError 1
J-Link Commander will now exit on Error
J-Link>r
J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link V13 compiled Feb 20 2025 16:24:39
Hardware version: V13.00
J-Link uptime (since boot): 0d 01h 53m 05s
S/N: 53000571
License(s): GDB
USB speed mode: High speed (480 MBit/s)
VTref=3.321V
Target connection not established yet but required for command.
Device "STM32G0B1RE" selected.

Connecting to target via SWD
InitTarget() start
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
InitTarget() end - Took 10.5ms
Found SW-DP with ID 0x0BC11477
DPIDR: 0x0BC11477
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x04770031, ADDR: 0x00000000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xF0000000
CPUID register: 0x410CC601. Implementer code: 0x41 (ARM)
Found Cortex-M0 r0p1, Little endian.
FPUnit: 4 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ F0000000
Memory zones:
  Zone: "Default" Description: Default access mode
Cortex-M0 identified.
Reset delay: 0 ms
Reset type: NORMAL (https://wiki.segger.com/J-Link_Reset_Strategies)
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
J-Link>LE
J-Link>loadfile "build/app_cmb/zephyr/merged.hex"
'loadfile': Performing implicit reset & halt of MCU.
Reset type: NORMAL (https://wiki.segger.com/J-Link_Reset_Strategies)
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
Downloading file [build/app_cmb/zephyr/merged.hex]...

****** Error: Failed to preserve target RAM @ 0x20000000-0x2001FFFF.
Failed to prepare for programming.
Unspecified error -1

Script processing completed.

OnDisconnectTarget() start
OnDisconnectTarget() end - Took 770us
FATAL ERROR: command exited with status 1: /opt/SEGGER/JLink_V822/JLinkExe -nogui 1 -if swd -speed 4000 -device STM32G0B1RE -CommanderScript /tmp/tmpcwy6nphrjlink/runner.jlink -nogui 1
```

## Workaround

Use the STM32 cube instead of JLink using `-r stm32cubeprogrammer`

marc-hb, 2025-09-08:
Thanks @efra-mx-fire , `west` supports "extensions" and `west flash` is not part of `west`, it's part of Zephyr here: https://github.com/zephyrproject-rtos/zephyr/tree/main/scripts/west_commands

I'm not even sure this is an issue in Zephyr either, isn't this a problem with `/opt/SEGGER/JLink_V822/JLinkExe`? Which is definitely not part of Zephyr.

efra-mx-fire, 2025-09-09:
ok, i will open the issue in the Zephyr project.

I agree with you, I am not sure if it is Zephyr or not, but it could be that that some parameter is missing

closed 2025-11-23 by github-actions[bot]