zephyrproject-rtos/zephyr #95661: Planned USB host support work

jfischer-no, 2025-09-08:
### Problem Description

There is a problem.

### Proposed Change (Summary)

Nothing.

### Proposed Change (Detailed)

**Important**
_This is a meta issue for maintainers and developers of relevant subsystems to track progress on USB support. Please do not ask general questions or make comments about USB support here. Please do not start a metaphysical discussion about the reasons for the existence of the host stack._

### Problem Description

Follow up on #42066, focusing on host support. For the USB device support see #95649.

Support for USB hosts in Zephyr is ramping up, but it will take some time to reach the same level as support for devices. **It is not ready for productive use. We cannot say when it will be ready, nor can we specify an exact date for the implementation of feature XYZ.** Please note that all APIs are experimental and may change at any time. Bear this in mind when working on new drivers or class drivers.

There are basic tests for both USB device and host support, which run in CI on the PR basis. We also have experimental USB/IP support, which is primarily used for exporting USB devices. There should not be any regressions, as these would also affect users of the USB device stack.

## Drivers

If you start working on the new driver, please use [MAX3421E or virtual driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/uhc/uhc_max3421e.c) as a reference point and template for the new driver. Please follow the [coding style guidelines](https://docs.zephyrproject.org/latest/contribute/style/code.html) and keep the style consistent with the USB drivers/subsystem code. Please note that clangformat does not format code according to the coding style guidelines, so you will need to fix it afterwards.

  - [ ] New driver for controller foobar ()

## Stack

The stack is in an experimental, early-development state. Ongoing work:

  - [ ] Class (Driver) API (@josuah https://github.com/zephyrproject-rtos/zephyr/pull/94590)

## USB Class/Function driver

There is no support for any classes or functions in the Zephyr tree. However, there is ongoing work and draft PRs on some functions. There are also no examples available yet that can be used as templates. The plan is to first add a basic test function as a counterpart to the device's loopback function, which could then be used for testing and as a template for the new class driver implementations. If you want to start work on the new class/function driver, this is the right issue to discuss. Please follow the coding style guidelines and keep the style consistent with the USB drivers/subsystem code. Please note that clangformat does not format code according to the coding style guidelines, so you will need to fix it afterwards.

  - [ ] Implementation of USB classes support
      - [ ] testusb function support as loopback counterpart
      - [ ] CDC ACM Class support (WIP @jfischer-no)
      - [ ] USB Video class driver (WIP #93517 @AidenHu @josuah)
      - [ ] Audio Class 2 driver
      - [ ] HID class support
      - [ ] USB DFU class support
      - [ ] USB MSC class driver
      - [ ] Bluetooth HCI function driver
      - [ ] CDC ECM/NCM functions driver

MarkWangChinese, 2025-11-05:
Hi @jfischer-no NXP has customer requirement for the USB Host ECM and USB Host HUB, so we have started to implement them. If it is OK for you, I will edit the corresponding feature items as WIP. Please provide your comments, thanks.

nimams, 2025-12-16:
Hello @jfischer-no 
do you have any update on CDC-ACM host class?
I want to use for a direct access modem to a usb modem and I was wondering how far you are on that.

MarkWangChinese, 2026-04-10:
> Hello we (NXP) have a customer waiting for the USB Host Audio implementation, but to be honest I am not sure if there is any activity planned already or some job done. Could you please keep me updated soon about this. Because we need to inform the customer and accept the adequate actions. Thanks, Michael

@AidenHu start to work on enabling the host audio2.0, so I add @AidenHu name to the task item. @jfischer-no

maass-hamburg, 2026-09-23:
For usb host controllers with multiple ports, do we want them to emulate a usb hub as the root hub, like linux, windows, freebsd and uboot do? or do something else?

I added usb hub emulation to my usb ohci controller:
https://github.com/zephyrproject-rtos/zephyr/pull/107065

maass-hamburg, 2026-09-23:
For using the devicetree for the usb devices that are connected to a usb host controller, there is president in the linux kernel: https://github.com/torvalds/linux/blob/fe2ec83746e501645709761605c2464a44fd2929/Documentation/devicetree/bindings/usb/usb-device.yaml and http://www.devicetree.org/open-firmware/bindings/usb/usb-1_0.ps
But in Linux they usually use it for hard wired usb devices. And I think we should at least support it for fixed usb devices like linux does. But that requires the possibility to bind a specific class instance to a specific port of the usb host controller or a usb hub.

thenguyenyf, 2026-09-29:
Hello. I'd want to inform that we have sent out the support for USBH DFU and Billboard class driver 
- https://github.com/zephyrproject-rtos/zephyr/pull/119550
- https://github.com/zephyrproject-rtos/zephyr/pull/119506

For the USBH HID support, we have sent out a PR that adding both HID boot and HID report protocol. However, an existing PR for HID boot has been sent out by @fkokosinski. We are discussing how the review process and feature alignment for both PRs could move forward. Sorry for the delayed information, which could result in duplication of development efforts.
- https://github.com/zephyrproject-rtos/zephyr/pull/119531

We have also started developing a vendor-specific class driver. It is ready locally, and the pull requests will be sent out shortly.

@jfischer-no , @josuah . If there is no issue with you, I'd like to update these items into planned list.

D-Veda, 2026-11-04:
Hello, I've already implemented the basic 1.0 version of NCM here. I'm still organizing the code, and the support for NCM 1.1 will be considered in the future.

- https://github.com/zephyrproject-rtos/zephyr/pull/124485

It is verified in frdm_mcxn236, refer https://github.com/zephyrproject-rtos/zephyr/pull/124306
@jfischer-no , @josuah If possible, I'd like to update it to the plan list