zephyrproject-rtos/zephyr #95729: rp2040: Use nvs fail. the prompt is written successfully, but cannot be read

zhang-wenchao, 2025-09-09:
### Describe the bug

### Add overlay

`samples/nvs/boards` add pico overlay

`rpi_pico.overlay`
```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		code_partition: partition@100 {
			label = "code";
			reg = <0x100 (DT_SIZE_M(1) - 0x100)>;
			read-only;
		};

		storage_partition: partition@100000 {
			label = "storage";
			reg = <0x100000  DT_SIZE_M(1)>;
		};
	};
};

```

### Erase flash
```
 /usr/local/bin/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "init; reset init; flash erase_sector 0 0 last; reset; exit"
```

### Build and flash

```
west build -b rpi_pico samples/subsys/nvs -- -DRPI_PICO_DEBUG_ADAPTER=cmsis-dap -DOPENOCD=/usr/local/bin/openocd -DOPENOCD_DEFAULT_PATH=/usr/local/share/openocd/scripts && west flash
```

### Trace Log

```
*** Booting Zephyr OS build v4.2.0-2084-g0fe091bc4691 ***
[00:00:00.003,000] <dbg> fs_nvs: nvs_recover_last_ate: Recovering last ate from sector 0
[00:00:00.016,000] <dbg> fs_nvs: nvs_add_gc_done_ate: Adding gc done ate at ff0
[00:00:00.044,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.048,000] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:00.053,000] <inf> fs_nvs: data wra: 0, 0
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
Reboot counter history: 
 Error, no Reboot counter
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build v4.2.0-2084-g0fe091bc4691 ***
[00:00:00.003,000] <dbg> fs_nvs: nvs_recover_last_ate: Recovering last ate from sector 0
[00:00:00.016,000] <dbg> fs_nvs: nvs_add_gc_done_ate: Adding gc done ate at ff0
[00:00:00.048,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.051,000] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:00.056,000] <inf> fs_nvs: data wra: 0, 0
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
Reboot counter history: 
 Error, no Reboot counter
[... 425 lines cut ...]
[00:00:00.046,000] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:00.052,000] <inf> fs_nvs: data wra: 0, 0
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
Reboot counter history: 
 Error, no Reboot counter
Rebooting in ...5...4...3...2...1
```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

zhang-wenchao, 2025-09-09:
@soburi You should be an expert on pico. Do you have time to look into this? I'm sorry to bother you, thank you very much for your time.

Laczen, 2025-09-10:
@zhang-wenchao, it seems that nothing is written to flash at all. Please enable flash debugging and rerun the test (the output can be limited to only one reboot).

drensber, 2025-09-18:
I am seeing a similar problem. My application uses NVS. It previously worked, but now it can only read NVS items that have already been written (it cannot erase them or overwrite new values to them). I've determined that changeset "6e3a1fda493a63806a315f56b57959ba57a2c9ba : manifest: hal_rpi_pico: Pull partial flash write changes from hal_rpi_pico" us the one where things broke on the Pico W (I haven't tried Pico 2 W yet). https://github.com/zephyrproject-rtos/zephyr/commit/6e3a1fda493a63806a315f56b57959ba57a2c9ba

TerryGeng, 2025-11-18:
I went back to the last commit before https://github.com/zephyrproject-rtos/zephyr/pull/89182 and tried to reproduce whatever was wrong. I figured: the moment I move the `flash_write_partial` function and its dependencies from `flash_rpi_pico.c` into the `hardware_flash/flash.c`, the flash write starts to fail silently. This doesn't make sense to me at all...