zephyrproject-rtos/zephyr #95755: BUILD_ASSERT in net_if.h causes clang-tidy to fail compilation

MichaelFeistETC, 2025-09-09:
### Describe the bug

Running `clang-tidy` on a file including `zephyr/include/zephyr/net/net_if.h` causes `clang-tidy` to fail compilation of the file. `clang-tidy` hits this line `BUILD_ASSERT((enum net_if_oper_state)(-1) > 0 && NET_IF_OPER_UNKNOWN == 0);` near line 915 and spits out the following build error:

```
zephyr/include/zephyr/net/net_if.h:915:15: error: static assertion expression is not an integral constant expression [clang-diagnostic-error]
  915 |         BUILD_ASSERT((enum net_if_oper_state)(-1) > 0 && NET_IF_OPER_UNKNOWN == 0);
      |                      ^
zephyr/include/zephyr/toolchain/gcc.h:76:50: note: expanded from macro 'BUILD_ASSERT'
   76 | #define BUILD_ASSERT(EXPR, MSG...) static_assert(EXPR, "" MSG)
      |                                                  ^~~~
zephyr/include/zephyr/net/net_if.h:915:15: note: integer value -1 is outside the valid range of values [0, 7] for the enumeration type 'net_if_oper_state'
  915 |         BUILD_ASSERT((enum net_if_oper_state)(-1) > 0 && NET_IF_OPER_UNKNOWN == 0);
```

This stops me from running `clang-tidy` on my own code.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Run clang-tidy on a file including net_if.h.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

Present on bb699ff2df249addb9473d0bbd2f6949f008161e

### Additional Context

I'd be happy to make a fix if I could better understand what the -1 check is actually trying to get at. Is it trying to assert that the enum is unsigned?

MichaelFeistETC, 2025-09-09:
Seems you used to be able to disable this behavior in clang in an earlier version and this behavior for clang isn't going away:
https://reviews.llvm.org/D131307
https://github.com/llvm/llvm-project/commit/cc104113ddecbdcec2cca848adbb6afa1214e9de

rlubos, 2025-09-10:
> Is it trying to assert that the enum is unsigned?

That was the case if I remember correctly, as a solution (or workaround?) for some other overly protective compiler warning.

Perhaps we could move the static assert to the `net_if.c`? It shouldn't matter where the build assert is placed. If that doesn't work though, I'd say just remove it.

MichaelFeistETC, 2025-09-10:
> That was the case if I remember correctly, as a solution (or workaround?) for some other overly protective compiler warning.
>
> Perhaps we could move the static assert to the `net_if.c`? It shouldn't [...]

Sounds like a good idea to me as long as there aren't any concerns with the source being able to be compiled by clang. At least for my use case, that solves my problem and is a simple fix.

closed 2025-09-15 by nashif