zephyrproject-rtos/zephyr #95782: doc: Documentation issue in 'boards/raspberrypi/index'

JamesH65, 2025-09-10:
### Describe the bug

Its Raspberry Pi Ltd, not Foundation.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Page: boards/raspberrypi/index
- Version: main
- SHA-1: 8dc3f856229ce083c956aa301c31a23e65bd8cd8

### Additional Context

This issue was reported from the online documentation page using the 'Report an issue' button.

soburi, 2025-09-16:
@JamesH65 

If the Foundation only manages the finances and the actual development is done by 'Raspberry Pi Limited', I think it would be better to change it, but it would also be preferable to have it be the same as the Linux setting. Are there any similar discussions on Linux?

soburi, 2025-09-16:
@JamesH65 

Thank you for explanation, and I got it the context.
#96081 is enough reasonable and acceptable to change.

closed 2025-11-30 by github-actions[bot]